ros-visualization/rqt #100: rqt_cpp does not work with tf

garaemon, 2015-02-08:
Cannot include tf/transform_listener.h from rqt cpp programs.
[This repository](https://github.com/garaemon/rqt_cpp_tf) is the sample project and it produces following error:
```
[ 66%] [100%] Building CXX object CMakeFiles/rqt_cpp_tf.dir/moc_my_plugin.cxx.o
Building CXX object CMakeFiles/rqt_cpp_tf.dir/my_plugin.cpp.o
In file included from /usr/include/boost/signals/connection.hpp:13:0,
                 from /usr/include/boost/signals/signal_template.hpp:18,
                 from /usr/include/boost/signals/signal0.hpp:24,
                 from /usr/include/boost/signal.hpp:19,
                 from /usr/include/boost/signals.hpp:9,
                 from /opt/ros/hydro/include/tf/tf.h:45,
                 from /opt/ros/hydro/include/tf/transform_listener.h:38,
                 from /home/garaemon/ros/hydro/build/rqt_cpp_tf/../../src/rqt_c
pp_tf/my_plugin.h:6,
                 from /home/garaemon/ros/hydro/build/rqt_cpp_tf/moc_my_plugin.c
xx:10:
/usr/include/boost/signals/detail/signals_common.hpp:26:13: error: expected ide
ntifier before protected
/usr/include/boost/signals/detail/signals_common.hpp:26:13: error: expected unq
ualified-id before protected
```

dirk-thomas, 2015-02-08:
I just tried to build the provided repository and it compiled and linked without a problem.

garaemon, 2015-02-08:
wow really? Do you use indigo?

dirk-thomas, 2015-02-08:
Yes

garaemon, 2015-02-08:
It might be problem only on hydro... I will check it on indigo too.

garaemon, 2015-02-08:
I confirmed that it worked on indigo. Hmm...

garaemon, 2015-02-08:
It looks like boost/qt related problem. Is there any maintainer who knows how to avoid this error?

dirk-thomas, 2015-02-09:
The signals/slots of boost and Qt seem to collide: http://www.boost.org/doc/libs/1_57_0/doc/html/signals/s04.html#idp426500832

Since you can neither change the rqt plugin headers using Qt signals / slot nor the tf headers using boost I am not sure which workaround will work for you.

dirk-thomas, 2015-04-30:
Since this is not a bug in rqt itself there is little I can do about it. Since it has been addressed upstream in newer versions I will close this (marked as wontfix). If anyone has input how to workaround this in rqt please comment to reopen the ticket.

closed 2015-04-30 by dirk-thomas