ros-visualization/rqt #101: No publisher plugin available for RQT

chrisspen, 2015-05-03:
Under the "Use RQT To Send Commands" section of the [ROS control tutorial](http://gazebosim.org/tutorials/?tut=ros_control) it says:

    Add the 'Publisher' plugin then choose the topic from the drop down box that commands any particular controller that you want to publish to.

However, RQT no longer displays any such menu item. I've installed all the `ros-indigo-rqt-*` packages on Ubuntu 14.04, and [this is a screenshot](http://i.imgur.com/IFOv3Pp.png) of the resulting RQT plugins menu. There's no "Topics->Publisher" item available.

I originally reported this as a bug to the [gazebo_ros_demos](https://github.com/ros-simulation/gazebo_ros_demos/issues/9#issuecomment-98445052) maintainer, but he claims this is a problem with RQT.

dirk-thomas, 2015-05-03:
The plugin is called "Message Publisher" in the groups "Topics" in the "Plugins" menu. It should be available when you install `ros-indigo-rqt-publisher`. I would assume that this is not a bug in rqt itself but a problem with your specific installation because it works well for others.

Can you please post more information about your installation, e.g. the output of the following command:

    dpkg -l | grep ros-indigo-rqt

chrisspen, 2015-05-03:
I have ros-indigo-rqt-publisher installed as well as every other ros-indigo-rqt-* package.

    ii  ros-indigo-rqt                                              0.2.14-1trusty-20150327-0415-+0000                     amd64        rqt is a Qt-based framework for GUI development for ROS. It consists of three parts/metapackages rqt (you're here) rqt_common_plugins - ROS backend tools suite that can be used on/off of robot runtime. rqt_robot_plugins - Tools for interacting with robots during their runtime. rqt metapackage provides a widget rqt_gui that enables multiple `rqt` widgets to be docked in a single window.
    ii  ros-indigo-rqt-action                                       0.3.10-0trusty-20150329-1114-+0000                     amd64        rqt_action provides a feature to introspect all available ROS action (from actionlib) types. By utilizing rqt_msg, the output format is unified with it and rqt_srv. Note that the actions shown on this plugin is the ones that are stored on your machine, not on the ROS core your rqt instance connects to.
    ii  ros-indigo-rqt-bag                                          0.3.10-0trusty-20150326-1826-+0000                     amd64        rqt_bag provides a GUI plugin for displaying and replaying ROS bag files.
    ii  ros-indigo-rqt-bag-plugins                                  0.3.10-0trusty-20150326-1829-+0000                     amd64        rqt_bag provides a GUI plugin for displaying and replaying ROS bag files.
    ii  ros-indigo-rqt-capabilities                                 0.1.2-0trusty-20150327-0555-+0000                      amd64        rqt package for visualization and management of capabilities
    ii  ros-indigo-rqt-common-plugins                               0.3.10-0trusty-20150329-1117-+0000                     amd64        rqt_common_plugins metapackage provides ROS backend graphical tools suite that can be used on/off of robot runtime. To run any rqt plugins, just type in a single command &quot;rqt&quot;, then select any plugins you want from the GUI that launches afterwards. rqt consists of three following metapackages: rqt - core modules of rqt (ROS GUI) framework. rqt plugin developers barely needs to pay attention to this metapackage. rqt_common_plugins (you're here!) rqt_robot_plugins - rqt plugins that are particularly used with robots during their runtime.
    ii  ros-indigo-rqt-console                                      0.3.10-0trusty-20150327-0621-+0000                     amd64        rqt_console provides a GUI plugin for displaying and filtering ROS messages.
    ii  ros-indigo-rqt-controller-manager                           0.9.1-0trusty-20150328-0536-+0000                      amd64        The rqt_controller_manager package
    ii  ros-indigo-rqt-dep                                          0.3.10-0trusty-20150327-0619-+0000                     amd64        rqt_dep provides a GUI plugin for visualizing the ROS dependency graph.
    ii  ros-indigo-rqt-ez-publisher                                 0.3.0-0trusty-20150328-2202-+0000                      amd64        The rqt_ez_publisher package
    ii  ros-indigo-rqt-graph                                        0.3.10-0trusty-20150327-0553-+0000                     amd64        rqt_graph provides a GUI plugin for visualizing the ROS computation graph. Its components are made generic so that other packages where you want to achieve graph representation can depend upon this pkg (use rqt_dep to find out the pkgs that depend. rqt_dep itself depends on rqt_graph too).
    ii  ros-indigo-rqt-graphprofiler                                0.1.2-0trusty-20150328-2119-+0000                      amd64        An experimental visualization system for anonymous publish subscribe architectures.
    ii  ros-indigo-rqt-gui                                          0.2.14-1trusty-20150326-1724-+0000                     amd64        rqt_gui provides the main to start an instance of the ROS integrated graphical user interface provided by qt_gui.
    ii  ros-indigo-rqt-gui-cpp                                      0.2.14-1trusty-20150326-1722-+0000                     amd64        rqt_gui_cpp enables GUI plugins to use the C++ client library for ROS.
    ii  ros-indigo-rqt-gui-py                                       0.2.14-1trusty-20150326-1729-+0000                     amd64        rqt_gui_py enables GUI plugins to use the Python client library for ROS.
    ii  ros-indigo-rqt-image-view                                   0.3.10-0trusty-20150326-1825-+0000                     amd64        rqt_image_view provides a GUI plugin for displaying images using image_transport.
    ii  ros-indigo-rqt-joint-trajectory-controller                  0.9.1-0trusty-20150326-1743-+0000                      amd64        Graphical frontend for interacting with joint_trajectory_controller instances.
    ii  ros-indigo-rqt-launch                                       0.3.10-0trusty-20150328-0652-+0000                     amd64        This rqt plugin ROS package provides easy view of .launch files. User can also start and end node by node that are defined in those files.
    ii  ros-indigo-rqt-logger-level                                 0.3.10-0trusty-20150327-0554-+0000                     amd64        rqt_logger_level provides a GUI plugin for configuring the logger level of ROS nodes. rqt_logger_level takes over `wx`-based tool [[rxloggerlevel]].
[... cut ...]
    ii  ros-indigo-rqt-topic                                        0.3.10-0trusty-20150326-1844-+0000                     amd64        rqt_topic provides a GUI plugin for displaying debug information about ROS topics including publishers, subscribers, publishing rate, and ROS Messages.
    ii  ros-indigo-rqt-web                                          0.3.10-0trusty-20150326-1745-+0000                     amd64        rqt_web is a simple web content viewer for rqt. Users can show web content in Qt-based window by specifying its URL.

ablasdel, 2015-05-03:
Mine seems fine but it seems to be a newer build of the same release

mine:
```
ii  ros-indigo-rqt-publisher                              0.3.10-0trusty-20150429-2357-+0000                  amd64        rqt_publisher provides a GUI plugin for publishing arbitrary messages with fixed or computed field values.
```
chrisspen's:
```
ii  ros-indigo-rqt-publisher                                    0.3.10-0trusty-20150327-0549-+0000                     amd64        rqt_publisher provides a GUI plugin for publishing arbitrary messages with fixed or computed field values.
```

@chrisspen try:
```
sudo apt-get update && sudo apt-get upgrade ros-indigo-* -y
```

chrisspen, 2015-05-04:
This is odd. I just re-ran:

    sudo apt-get install ros-indigo-desktop-full

and it ran perfectly, but when I run the upgrade I get the error:

    The following packages have unmet dependencies:
      ros-indigo-ipa-canopen : Depends: ros-indigo-ipa-canopen-core but it is not installable
                          Depends: ros-indigo-ipa-canopen-ros but it is not installable

It seems neither of those packages exist in the repo...

dirk-thomas, 2015-05-04:
What does the following command output:

    rqt --list-plugins

Please also try running and post the output:

    rospack profile

chrisspen, 2015-05-04:
Ah, thanks. It looks like my reinstall of desktop-full fixed the RQT plugin issue, because I'm now seeing the Topic item. This issue with upgrading and `ros-indigo-ipa-canopen` seems unrelated, so I'll report that elsewhere.

closed 2015-05-04 by chrisspen

davetcoleman, 2015-05-04:
Glad we got to the bottom of this!

Chrispako990210, 2024-10-10:
On a side note, I had the same problem with ros humble. Ive done sudo apt install ros-humble-rqt-* and reinstalled the desktop-full. When I use the dpkg -l I see the packages downloaded but with rqt --list-plugins they dont show up. What worked for me was to launch rqt --force-discovery. Does anybody know how to work around this?