ros-visualization/rqt #102: export perspective opens a blank window

CatherineH, 2015-05-19:
When I try to export my perspective, I get a blank window. Here's a screenshot:

[image]

I'm using Ubuntu 14.04, and I installed rqt_gui from ros-indigo-desktop-full.

dirk-thomas, 2015-09-17:
The code is only invoking `QFileDialog.getSaveFileName` (https://github.com/ros-visualization/qt_gui_core/blob/22f5c031ae97ba9ad88c0974f8dbf7bd2ae6b8ea/qt_gui/src/qt_gui/perspective_manager.py#L346). Therefore I don't think this can be a bug in rqt but more likely a rendering glitch in Qt itself.

Do you still have this problem on an up-to-date system?

shashwatak, 2015-10-16:
Hi, I am having an identical problem, any word on what this is or how to resolve it?

Is there another way to export perspectives?

dirk-thomas, 2016-09-08:
I will close this ticket and mark it with "wontfix" since `rqt` can't do anything to address this afaik.

closed 2016-09-08 by dirk-thomas