ros-visualization/rqt #103: Improper Rendering of text in rqt

sidcode, 2015-05-30:
The attached figure - [image]

ROS Indigo Version 1.11.13
Ubuntu 14.04
rqt_tf_tree  0.4.1
rqt_graph 0.3.11

The problem of Text rendering persists in all rqt applications for me.

dirk-thomas, 2015-05-30:
Please reopen the ticket in the issue tracker of the repository where `rqt_tf_tree` is in (https://github.com/ros-visualization/rqt_robot_plugins) and close this.

If a similar problem exists in `rqt_graph` please consider filling a separate issue in https://github.com/ros-visualization/rqt_common_plugins (with its own screenshot).

sidcode, 2015-05-30:
Thanks, I'll do that.

closed 2015-05-30 by sidcode