ros-visualization/rqt #104: font size too large in rqt_graph, rqt_tf_tree etc. on high-dpi screens

jmfurrer, 2015-06-01:
as reported [here on answers.ros.org](http://answers.ros.org/question/190133/rqt_graph-font-too-large/) about a year ago

dirk-thomas, 2015-06-01:
Please fill the ticket in the repository which contains the package and close this one:

* rqt_graph: https://github.com/ros-visualization/rqt_common_plugins
* rqt_tf_tree: https://github.com/ros-visualization/rqt_robot_plugins (the ticket https://github.com/ros-visualization/rqt_robot_plugins/issues/85 might be a duplicate?)

jmfurrer, 2015-06-01:
Had put it here as it seemed to be a general rqt problem (affects rqt_dep, rqt_graph, rqt_tf_tree...). Closed as issue ros-visualization/rqt_robot_plugins#85 describes this already.

closed 2015-06-01 by jmfurrer