ros-visualization/rqt #105: PySide image not found

SorJEF, 2016-03-13:
I'm using OS X El Capitan with the ROS Jade installed. When trying to run rqt I get the following error:

```
Traceback (most recent call last):
  File "/Users/artemsorokin/ros/install_isolated/bin/rqt", line 13, in <module>
    sys.exit(main.main())
  File "/Users/artemsorokin/ros/install_isolated/lib/python2.7/site-packages/rqt_gui/main.py", line 59, in main
    return super(Main, self).main(argv, standalone=standalone, plugin_argument_provider=plugin_argument_provider, plugin_manager_settings_prefix=str(hash(os.environ['ROS_PACKAGE_PATH'])))
  File "/Users/artemsorokin/ros/install_isolated/lib/python2.7/site-packages/qt_gui/main.py", line 336, in main
    from python_qt_binding import QT_BINDING
  File "/Users/artemsorokin/ros/build_isolated/python_qt_binding/bdist.macosx-10.11-x86_64/egg/python_qt_binding/__init__.py", line 55, in <module>
  File "/Users/artemsorokin/ros/build_isolated/python_qt_binding/bdist.macosx-10.11-x86_64/egg/python_qt_binding/binding_helper.py", line 267, in <module>
  File "/Users/artemsorokin/ros/build_isolated/python_qt_binding/bdist.macosx-10.11-x86_64/egg/python_qt_binding/binding_helper.py", line 95, in _select_qt_binding
ImportError: Could not find Qt binding (looked for: 'pyqt', 'pyside'):
  ImportError for 'pyqt': dlopen(/usr/local/lib/python2.7/site-packages/sip.so, 2): Symbol not found: _PyUnicodeUCS2_AsASCIIString
  Referenced from: /usr/local/lib/python2.7/site-packages/sip.so
  Expected in: flat namespace
 in /usr/local/lib/python2.7/site-packages/sip.so
  ImportError for 'pyside': dlopen(/usr/local/var/pyenv/versions/2.7.11/lib/python2.7/site-packages/PySide/QtCore.so, 2): Library not loaded: @rpath/libpyside-python2.7.1.2.dylib
  Referenced from: /usr/local/var/pyenv/versions/2.7.11/lib/python2.7/site-packages/PySide/QtCore.so
  Reason: image not found

```

SorJEF, 2016-03-13:
Ok, this is related to https://github.com/PySide/PySide/issues/129

The temporary workaround with `DYLD_LIBRARY_PATH` described in that ticket worked. So if you observe this issue try executing this command and adding it to your `.bash_profile` (on OSX):
    
    export DYLD_LIBRARY_PATH=/Users/you/virtualenv/lib/python2.7/site-packages/PySide

dirk-thomas, 2016-03-30:
I will close this ticket since it is an upstream problem and there is nothing `rqt` can do.

closed 2016-03-30 by dirk-thomas