ros-visualization/rqt #107: Kinetic rqt dependency missing

theelectricwiz, 2016-05-07:
When running `rosrun rqt_graph rqt_graph` on a fresh install of ROS Kinetic (`ros-kinetic-desktop-full/xenial,now 1.3.0-0xenial-20160429-172040-0700 amd64`) and Ubuntu 16.04 LTS, I get the following error due to a missing dependency:

```
RosPluginProvider.load(rqt_graph/RosGraph) exception raised in __builtin__.__import__(rqt_graph.ros_graph, [RosGraph]):
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui/ros_plugin_provider.py", line 77, in load
    module = __builtin__.__import__(attributes['module_name'], fromlist=[attributes['class_from_class_type']], level=0)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_graph/ros_graph.py", line 39, in <module>
    from python_qt_binding.QtSvg import QSvgGenerator
ImportError: No module named QtSvg

PluginManager._load_plugin() could not load plugin "rqt_graph/RosGraph":
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/plugin_handler.py", line 99, in load
    self._load()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/plugin_handler_direct.py", line 54, in _load
    self._plugin = self._plugin_provider.load(self._instance_id.plugin_id, self._context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui_py/ros_py_plugin_provider.py", line 60, in load
    return super(RosPyPluginProvider, self).load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui/ros_plugin_provider.py", line 83, in load
    raise e
ImportError: No module named QtSvg
```

Fixed by installing package `python-pyqt5.qtsvg` (default version of Python - `Python 2.7.11+`)

dirk-thomas, 2016-05-07:
Duplicate of ros-visualization/rqt_common_plugins#373

closed 2016-05-07 by dirk-thomas

theelectricwiz, 2016-05-07:
Ah. Apologies @dirk-thomas