ros-visualization/rqt #109: rqt plot (Kinetic) crashes in Ubuntu 16.04

ChristophTobler, 2016-09-05:
If I open rqt with `rqt` I can use image view etc. But if I try to open a plot, nothing happens or I get: Segmentation fault (core dumped) 
When I run `rosrun rqt_plot rqt_plot` I get the following error:

Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/rqt_plot/rqt_plot", line 6, in <module>
    from rqt_plot.plot import Plot
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_plot/plot.py", line 43, in <module>
    from .data_plot import DataPlot
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_plot/data_plot/__init__.py", line 44, in <module>
    from pyqtgraph_data_plot import PyQtGraphDataPlot
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_plot/data_plot/pyqtgraph_data_plot.py", line 37, in <module>
    from pyqtgraph import PlotWidget, mkPen, mkBrush
  File "/usr/lib/python2.7/dist-packages/pyqtgraph/__init__.py", line 13, in <module>
    from .Qt import QtGui
  File "/usr/lib/python2.7/dist-packages/pyqtgraph/Qt.py", line 104, in <module>
    from PyQt4 import QtGui, QtCore, uic
RuntimeError: the PyQt4.QtCore and PyQt5.QtCore modules both wrap the QObject class

This happened on both computers with Ubuntu 16.04 and Kinetic (desktop full)

Is there a way I can force plot to use just one version?

dirk-thomas, 2016-09-06:
It looks like you are trynig to use the `PyQtGraph` backend for `rqt_plot`. The standard installation should only provide the `Matplot` backend afaik. Can you please clarify how you have installed it? Did you manually install `python-pyqtgraph` (which is Qt 4 only)?

I don't think a Qt 5 version of PyQtGraph exists at the moment. I created ros-visualization/rqt_common_plugins#399 in the case `python-pyqtgraph` is installed manually on the system.

dirk-thomas, 2016-09-06:
Until the patch is merged and a new version of the package released you can apply the patch from the PR to your local files.

closed 2016-09-06 by dirk-thomas

ChristophTobler, 2016-09-07:
Thanks. Yes, I installed `python-pyqtgraph` manually because an other package uses it. 
So if I remove all rqt stuff (`sudo apt-get remove ros-kinetic-rqt-*` ?) and install it from source, it should work? Or will it just raise the error: 'PyQtGraph does not support Qt 5 at the moment' ?

dirk-thomas, 2016-09-07:
I wouldn't recommend installing all of rqt from source. Just apply the patch to the installed files. Whenever a new Debian package becomes available it will overwrite your modifications but the next patch releases should contain the fix anyway.

The plot plugin has three different backends (clock on the gear icon at the top right of the plot plugin to see). It should fall back to matplot.

zaidpirwani, 2017-01-25:
How can I "just apply the patch to the installed files"

please explain a bit, I dont know how?

dirk-thomas, 2017-01-25:
@zaidpirwani You have multiple options:

* Clone the `rqt_common_plugins` repo into your catkin workspace and use it from there.

* Modify the installed files in-place: edit `/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_plot/data_plot/pyqtgraph_data_plot.py` and apply the same diff as in ros-visualization/rqt_common_plugins#415.

* Use the [shadow-fixed](http://wiki.ros.org/ShadowRepository) / testing apt repo (instead of the official stable one) which already contains the new version 0.4.4 of `rqt_plot`: http://repositories.ros.org/status_page/ros_kinetic_default.html?q=rqt_plot

* Wait a little bit until the new version of the Debian package has propagated to the main apt repo. This step will also be announced, like this: https://discourse.ros.org/t/new-packages-for-jade-and-kinetic-2016-12-15/975

zaidpirwani, 2017-02-03:
Thanks, I think I will wait.

beetleskin, 2017-04-05:
With the currently released verision of python-pyqtgraph (0.9.10-5), this is still an issue. Or again? Running `rqt_plot` triggers:
```
Traceback (most recent call last):
  File "/opt/ros/kinetic/bin/rqt_plot", line 6, in <module>
    from rqt_plot.plot import Plot
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_plot/plot.py", line 43, in <module>
    from .data_plot import DataPlot
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_plot/data_plot/__init__.py", line 44, in <module>
    from .pyqtgraph_data_plot import PyQtGraphDataPlot
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_plot/data_plot/pyqtgraph_data_plot.py", line 46, in <module>
    from pyqtgraph import __version__ as pyqtgraph_version
  File "/usr/lib/python2.7/dist-packages/pyqtgraph/__init__.py", line 13, in <module>
    from .Qt import QtGui
  File "/usr/lib/python2.7/dist-packages/pyqtgraph/Qt.py", line 104, in <module>
    from PyQt4 import QtGui, QtCore, uic
RuntimeError: the PyQt4.QtCore and PyQt5.QtCore modules both wrap the QObject class
```

Diving into the code reveals that
```
from pyqtgraph import __version__ as pyqtgraph_version
```
already throws in the Qt-version mismatch exception. The actual raise beyond (`ImportError('A newer PyQtGraph version is required (at least 0.10 for Qt 5)')`) is never reached.

dirk-thomas, 2017-04-05:
@beetleskin Please provide more information which versions of the involved packages you are using.

Without being able to reproduce the problem you will need to debug further and identify why it is failing for you. You might want to try running the import commands in a Python interpreter to narrow down what exactly fails and why. One note which might help: in Kinetic `rqt` uses Qt 5 (not Qt 4).

vincentrou, 2017-04-07:
@beetleskin I had the same problem when installing pyqtgraph with apt (version 0.9.10)
But it works fine when using the deb package here http://www.pyqtgraph.org/

beetleskin, 2017-04-19:
@vincentrou True that .. so I guess we just wait until this one is released.

@dirk-thomas Reproduction is easy, just install the current releases of the two packages on Ubuntu 16.04:
* ros-kinetic-rqt-plot: 0.4.7-0xenial-20170308-221732-0800
* python-pyqtgraph: 0.9.10-5
.. and run `rqt_plot`

dirk-thomas, 2017-04-20:
@beetleskin `python-pyqtgraph` is using Qt 4. `rqt` in `Kinetic` is using Qt 5. So that combination can't work - hence the Qt version mismatch. You need at least version 0.10 of `pyqtgraph` (see [changelog](https://github.com/pyqtgraph/pyqtgraph/blob/develop/CHANGELOG)).

beetleskin, 2017-04-23:
Sure, I'm aware of the problem. But since there is already a Qt-version check and a fallback in rqt-plot, in case of unsupported qt-versions in pyqtgraph, and the currently stable released version of rqt_plot crahses with the currently stable released version of pyqtgraph .. why not fixing the issue and have a working version of rqt_plot in that (special?) case?
Manually installing another version of a library feels rather against all the effort the ros-ecosystems undertakes in order to have all-the-time stable versions of all the parts it consists of. I.e. right now I'd have to create my own rosdep linking the 0.10.x version of pyqtgraph in order to keep a working workspace..

dirk-thomas, 2017-04-23:
> why not fixing the issue and have a working version of rqt_plot in that (special?) case?

I might not understand your comment. Can you please clarify how you would suggest to fix the issue / propose a pull request to do so.