ros-visualization/rqt #110: Error when re-arranging widgets over plot. Kinetic, 16.04

vmatos, 2016-09-20:
When I try to re-arrange widgets in rqt, whenever I pass with it over a plot widget, I get the error:
```
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/dock_widget.py", line 90, in _event
    new_parent = self._get_new_parent(widget)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/dock_widget.py", line 139, in _get_new_parent
    widget = widget.parent()
TypeError: 'Canvas' object is not callable
```

vmatos, 2016-09-20:
If I don't got over the plot widget, and drop it in place, I get:

```
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_plot/data_plot/mat_data_plot.py", line 107, in resizeEvent
    self.figure.tight_layout()
  File "/usr/lib/python2.7/dist-packages/matplotlib/figure.py", line 1756, in tight_layout
    self.subplots_adjust(**kwargs)
  File "/usr/lib/python2.7/dist-packages/matplotlib/figure.py", line 1612, in subplots_adjust
    self.subplotpars.update(*args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/matplotlib/figure.py", line 230, in update
    raise ValueError('bottom cannot be >= top')
ValueError: bottom cannot be >= top
```

RYO0115, 2016-11-24:
I have the same problem running rqt_gui on Ubuntu 16.04.

> Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/dock_widget.py", line 90, in _event
    new_parent = self._get_new_parent(widget)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/dock_widget.py", line 139, in _get_new_parent
    widget = widget.parent()
TypeError: 'Canvas' object is not callable

> Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_plot/data_plot/mat_data_plot.py", line 107, in resizeEvent
    self.figure.tight_layout()
  File "/usr/lib/python2.7/dist-packages/matplotlib/figure.py", line 1756, in tight_layout
    self.subplots_adjust(**kwargs)
  File "/usr/lib/python2.7/dist-packages/matplotlib/figure.py", line 1612, in subplots_adjust
    self.subplotpars.update(*args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/matplotlib/figure.py", line 230, in update
    raise ValueError('bottom cannot be >= top')
ValueError: bottom cannot be >= top

I tried
 sudo apt-get install ros-kinetic-rqt-gui 
but, nothing changes.

dirk-thomas, 2016-12-20:
@vmatos @RYO0115 Can you please try the proposed fix from ros-visualization/qt_gui_core#76 and confirm if it fixes the problem for you.

If there are problems remaining can you please describe the exact steps how to reproduce it.

vmatos, 2016-12-21:
My procedure:
```
mkdir -p overlay_ws/src
cd overlay_ws/src
git clone https://github.com/ros-visualization/qt_gui_core.git -b fix_parent_logic
cd ..
catkin_make
```

Then I verified the error was reproducible:
1. overlaying workspace: `source devel/setup.bash`
1. run: `rqt`
2. Place 2 plot widgets.
3. Hold widget and try moving it over plot area.
4. Get error.

Then I tested the fix:
1. run: `rqt`
2. Place 2 plot widgets.
3. Hold widget and try moving it over plot area.
4. No error!

@dirk-thomas My test indicates this PR fixes my first issue: `TypeError: 'Canvas' object is not callable`

dirk-thomas, 2016-12-21:
I was able to reproduce the first error easily. The PR fixes (also for you) which is good. But I never saw the second error. What are the steps to reproduce the second one?

vmatos, 2016-12-21:
To reproduce the 2nd error:

1. run: `rqt`
2. Place 2 plot widgets. they should appear side by side.
3. Move one of the plot widgets over the other, so they are distributed vertically.
4. Get error

dirk-thomas, 2016-12-21:
I can reproduce the second error with these steps. Unfortunately the error is within `matplotlib` and I am not sure if there is anything we can do in the rqt plugin itself. Btw it is also ticketed upstream: matplotlib/matplotlib#5456