ros-visualization/rqt #111: rqt_* crash on KDE Neon 5.8

oKermorgant, 2016-11-23:
I use KDE Neon (based on Ubuntu with latest KDE stuff) since 16.04 without a flaw with ROS.
Since a recent upgrade, all rqt_ programs crash at startup. 

Did a quick install of Kubuntu 16.04 (without, then with KDE PPA) and in both cases everything runs fine. It seems to be related with Python-SIP and Qt version.

Here are the configurations :

On KDE Neon 5.8
KDE Plasma 5.8.4
Qt 5.7.0
Kernel 4.4.0.47-generic
any rqt_* gives error :  python: siplib.c:12299: sipEnumType_alloc: Assertion `sipTypeIsEnum(currentType)' failed.

On Kubuntu 16.04
KDE Plasma 5.5.5
Qt 5.5.1
Kernel 4.4.0.31-generic
-> ok

On Kubuntu 16.04 + PPA
KDE Plasma 5.6.5
Qt 5.5.1
Kernel 4.4.0.47-generic
-> ok

dirk-thomas, 2017-01-23:
Please consider to provide a PR since I don't have a platform to test this on.

oKermorgant, 2017-02-14:
I went back to classical 16.04, but still have a KDE Neon with Qt 5.7 in a virtual machine to run tests.
As of now all rqt_* programs still crash, but I guess this is due to issues in Python / Qt5. 

I do not have either the skills or the time to provide a PR, so I'll close this issue hoping that it will be solved by the PyQt5 guys eventually.

closed 2017-02-14 by oKermorgant