ros-visualization/rqt #112: rqt crashes due to import error on kinetic

igor-krawczuk, 2017-01-19:
Hope this is the right place to report this:

both `rosrun rqt_gui rqt_gui` and `rqt` give the following error on typing

Traceback (most recent call last):
  File "/opt/ros/kinetic/bin/rqt", line 13, in <module>
    sys.exit(main.main())
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui/main.py", line 59, in main
    return super(Main, self).main(argv, standalone=standalone, plugin_argument_provider=plugin_argument_provider, plugin_manager_settings_prefix=str(hash(os.environ['ROS_PACKAGE_PATH'])))
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/main.py", line 344, in main
    from .container_manager import ContainerManager
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/container_manager.py", line 33, in <module>
    from .dock_widget import DockWidget
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/dock_widget.py", line 33, in <module>
    from python_qt_binding.QtWidgets import QApplication, QDockWidget
ImportError: No module named QtWidgets
`
System context:

`$lsb_release -a:
 Distributor ID: Ubuntu
Description:    Ubuntu 16.04.1 LTS
Release:        16.04
Codename:       xenial
$rosversion -d
kinetic
$rosversion rqt_graph
0.4.3
$rosversion rqt_gui
0.3.1
`
ros-kinetic installed via official way, all seems to be up to date. I think this is due to the change from qt4 to qt5 if my googling gave me the right idea?

dirk-thomas, 2017-01-19:
Can you please also post the version number of `python_qt_binding`. Can I assume that you have installed these packages from the apt repo?

igor-krawczuk, 2017-01-19:
```
$rosversion python_qt_binding
0.3.1
```

yes, installed it all through the apt repo. I checked in the git repo of qt_gui_core in ros-visualizations though, and it does the same imports if I read it correctly.

dirk-thomas, 2017-01-20:
That is weird. Ir works fine for me as well as others. Lets try to debug a bit deeper. After sourcing the ROS environment can you please run the following commands in an interactive Python shell:

```
import python_qt_binding
print(python_qt_binding.QT_BINDING)
from python_qt_binding.QtWidgets import QWidget
```

The second line will tell us if you are using PyQt or PySide. The third line will likely fail since that is what rqt is calling in your stacktrace.

igor-krawczuk, 2017-01-23:
Here is the output

```
$ source /opt/ros/kinetic/setup.sh 
$ python
Python 2.7.12 (default, Nov 19 2016, 06:48:10) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import python_qt_binding
>>> print(python_qt_binding.QT_BINDING)
pyqt
>>> from python_qt_binding.QtWidgets import QWidget
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named QtWidgets
```

dirk-thomas, 2017-01-23:
It looks to me that you are using Qt4. Can you please additionally post the output of:

```
python_qt_binding.QT_BINDING_VERSION
python_qt_binding.QT_BINDING_MODULES
```

igor-krawczuk, 2017-01-23:
Seems so 
```
>>> python_qt_binding.QT_BINDING_VERSION
'4.11.4'
>>> python_qt_binding.QT_BINDING_MODULES
{'QtGui': <module 'PyQt4.QtGui' from '/usr/lib/python2.7/dist-packages/PyQt4/QtGui.so'>, 'QtXmlPatterns': <module 'PyQt4.QtXmlPatterns' from '/usr/lib/python2.7/dist-packages/PyQt4/QtXmlPatterns.so'>, 'QtDeclarative': <module 'PyQt4.QtDeclarative' from '/usr/lib/python2.7/dist-packages/PyQt4/QtDeclarative.so'>, 'QtSvg': <module 'PyQt4.QtSvg' from '/usr/lib/python2.7/dist-packages/PyQt4/QtSvg.so'>, 'QtScript': <module 'PyQt4.QtScript' from '/usr/lib/python2.7/dist-packages/PyQt4/QtScript.so'>, 'QtXml': <module 'PyQt4.QtXml' from '/usr/lib/python2.7/dist-packages/PyQt4/QtXml.so'>, 'QtWebKit': <module 'PyQt4.QtWebKit' from '/usr/lib/python2.7/dist-packages/PyQt4/QtWebKit.so'>, 'QtCore': <module 'PyQt4.QtCore' from '/usr/lib/python2.7/dist-packages/PyQt4/QtCore.so'>, 'QtNetwork': <module 'PyQt4.QtNetwork' from '/usr/lib/python2.7/dist-packages/PyQt4/QtNetwork.so'>}
```

Sh

dirk-thomas, 2017-01-23:
I am kind of puzzled. `python_qt_binding` in Kinetic only uses PyQt5 (https://github.com/ros-visualization/python_qt_binding/blob/c36afe8515c5424501f106fe6303cc84ec11e4df/src/python_qt_binding/binding_helper.py#L153). There is no code which would fall back to import PyQt.

Maybe you can add `print` statements to `/opt/ros/kinetic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py` to debug what is happening on your system.

igor-krawczuk, 2017-01-23:
like this? 
```
    print "should import pyqt5"
    import PyQt5.QtCore
    print PyQt5.QtCore.PYQT_VERSION_STR
    print "should have imported pyqt5"
    return PyQt5.QtCore.PYQT_VERSION_STR
```

This does not get printed on my system. Thank you for helping me debug with such an awesome response time by the way

dirk-thomas, 2017-01-23:
No problem, we have to figure it out somehow :wink:

Can you add a message before and after [this line](https://github.com/ros-visualization/python_qt_binding/blob/c36afe8515c5424501f106fe6303cc84ec11e4df/src/python_qt_binding/binding_helper.py#L89) as well as in the `else` and `except` block. The value of `QT_BINDING` at the beginning and end of that function might also be valuable.

igor-krawczuk, 2017-01-23:
Adding this:
```
def _select_qt_binding(binding_name=None, binding_order=None):
   global QT_BINDING, QT_BINDING_VERSION
   print QT_BINDING
   (....)
   for binding_name in binding_order:
        try:
            binding_loader = getattr(sys.modules[__name__], '_load_%s' % binding_name, None)
            if binding_loader:
                print "have binding_loader"
                QT_BINDING_VERSION = binding_loader(required_modules, optional_modules)
                print binding_name
                print "after QT_BINDING_VERSION"
                QT_BINDING = binding_name
                break
            else:
                print "before binding_loader error_msg_append"
                error_msgs.append("  Binding loader '_load_%s' not found." % binding_name)
                print "after binding_loader error_msg_append"
        except ImportError as e:
            print "in importerror binding_loader error_msg_append"
            error_msgs.append("  ImportError for '%s': %s\n%s" % (binding_name, e, traceback.format_exc()))
            print "after importerror binding_loader error_msg_append"

    if not QT_BINDING:
        raise ImportError("Could not find Qt binding (looked for: %s):\n%s" % (', '.join(["'%s'" % b for b in binding_order]), '\n'.join(error_msgs)))
    print QT_BINDING

```

leads to this output:
```
$ rqt
Traceback (most recent call last):
  File "/opt/ros/kinetic/bin/rqt", line 13, in <module>
    sys.exit(main.main())
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rqt_gui/main.py", line 59, in main
    return super(Main, self).main(argv, standalone=standalone, plugin_argument_provider=plugin_argument_provider, plugin_manager_settings_prefix=str(hash(os.environ['ROS_PACKAGE_PATH'])))
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/main.py", line 344, in main
    from .container_manager import ContainerManager
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/container_manager.py", line 33, in <module>
    from .dock_widget import DockWidget
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui/dock_widget.py", line 33, in <module>
    from python_qt_binding.QtWidgets import QApplication, QDockWidget
ImportError: No module named QtWidgets

```

dirk-thomas, 2017-01-23:
The other code still trips over the same problem. Can you keep your debug message but call the following `python_qt_binding` in a Python interpreter:

```
import python_qt_binding
```

That should print something... hopefully...

igor-krawczuk, 2017-01-23:
It did not in fact print something, so I got suspicios wether the python path is set correctly:
```
~/catkin_workspace$ source /opt/ros/kinetic/setup.sh 
~/catkin_workspace$ source devel/setup.sh 
~/catkin_workspace$ python
Python 2.7.12 (default, Nov 19 2016, 06:48:10) 
[GCC 5.4.0 20160609] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import python_qt_binding
>>> python_qt_binding.__file__
'/usr/lib/python2.7/dist-packages/python_qt_binding/__init__.pyc'
```

apparently not

dirk-thomas, 2017-01-23:
Very good catch :fishing_pole_and_fish:

Can you check with `dpkg -S /usr/lib/python2.7/dist-packages/python_qt_binding/__init__.py` which package provides this file. I fear it will be an upstream Debian package.

igor-krawczuk, 2017-01-23:
Seems so:
```
~/catkin_workspace$ dpkg -S /usr/lib/python2.7/dist-packages/python_qt_binding/__init__.py
python-qt-binding: /usr/lib/python2.7/dist-packages/python_qt_binding/__init__.py
```

should I try and remove and reinstall via pip?

dirk-thomas, 2017-01-23:
You should just purge this package (and maybe other packages?). You should already have `ros-kinetic-python-qt-binding` installed which provides the newer code.

In general on Ubuntu you shouldn't need to install anything via PIP.

Just as a reference: mixing Debian / Ubuntu upstream packages with the ROS packages provided by our apt repo is not working and has plenty of problems like this. See http://wiki.ros.org/UpstreamPackages for more information.

igor-krawczuk, 2017-01-24:
Thank you very much for the help in debugging this!

closed 2017-01-24 by igor-krawczuk

dirk-thomas, 2017-01-24:
Thanks for spending the time and helping debugging this!