ros-visualization/rqt #114: rqt segfaults on kinetic

juakofz, 2017-02-05:
Hi everybody. This is a question I originally posted on ROS answers ([link](http://answers.ros.org/question/253655/rqt-segmentation-fault-following-tutorials/))

I just installed ROS Kinetic Kame on Ubuntu 16.04 following the tutorials (full desktop install). When I get to the 6th beginner tutorial (Understanding ROS Topics, http://wiki.ros.org/ROS/Tutorials/Und... ), at 2.1, when I try to run rqt_graph, I get the message "Segmentation fault (core dumped)". This also happens when running rqt directly, or any other rqt package. This is the only package I have had trouble with, other packages like rviz seem to work perfectly.

I've been looking around for a solution, but I did not find anybody with this problem. I have tried reinstalling both ROS and rqt packages, but it didn't make any difference. I have updated Qt but nothing changed.
```
$ rosrun rqt_graph rqt_graph
Segmentation fault (core dumped)
```

```
$ gdb python
(gdb) run /opt/ros/kinetic/bin/rqt_graph
Starting program: /usr/bin/python /opt/ros/kinetic/bin/rqt_graph
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe732c700 (LWP 19956)]
[New Thread 0x7fffdd816700 (LWP 19957)]
[New Thread 0x7fffdd015700 (LWP 19958)]
[New Thread 0x7fffdc814700 (LWP 19959)]
[New Thread 0x7fffcf635700 (LWP 19961)]
[New Thread 0x7fffcee34700 (LWP 19962)]

Thread 1 "python" received signal SIGSEGV, Segmentation fault.
0x0000000000000000 in ?? ()

(gdb) backtrace
#0  0x0000000000000000 in ?? ()
#1  0x00007fffccf56337 in ?? ()
   from /opt/ros/kinetic/lib/python2.7/dist-packages/qt_gui_cpp/libqt_gui_cpp_sip.so
#2  0x00007fffef35e811 in ?? ()
   from /usr/lib/python2.7/dist-packages/sip.x86_64-linux-gnu.so
#3  0x00007fffef35b642 in ?? ()
   from /usr/lib/python2.7/dist-packages/sip.x86_64-linux-gnu.so
#4  0x00000000004b670c in ?? ()
#5  0x00000000004b0cb3 in PyObject_Call ()
#6  0x00000000004c9faf in PyEval_EvalFrameEx ()
#7  0x00000000004c2765 in PyEval_EvalCodeEx ()
#8  0x00000000004de6fe in ?? ()
#9  0x00000000004b0cb3 in PyObject_Call ()
#10 0x00000000004f492e in ?? ()
#11 0x00000000004b0cb3 in PyObject_Call ()
#12 0x00000000004f46a7 in ?? ()
#13 0x00000000004b670c in ?? ()
#14 0x00000000004b0cb3 in PyObject_Call ()
#15 0x00000000004c9faf in PyEval_EvalFrameEx ()
#16 0x00000000004c9d8f in PyEval_EvalFrameEx ()
#17 0x00000000004c9d8f in PyEval_EvalFrameEx ()
#18 0x00000000004c9d8f in PyEval_EvalFrameEx ()
#19 0x00000000004c9d8f in PyEval_EvalFrameEx ()
#20 0x00000000004c9d8f in PyEval_EvalFrameEx ()
#21 0x00000000004c9d8f in PyEval_EvalFrameEx ()
#22 0x00000000004c9d8f in PyEval_EvalFrameEx ()
#23 0x00000000004c2765 in PyEval_EvalCodeEx ()
#24 0x00000000004ca099 in PyEval_EvalFrameEx ()
#25 0x00000000004c2765 in PyEval_EvalCodeEx ()
#26 0x00000000004ca099 in PyEval_EvalFrameEx ()
#27 0x00000000004c2765 in PyEval_EvalCodeEx ()
#28 0x00000000004c2509 in PyEval_EvalCode ()
#29 0x00000000004f1def in ?? ()
#30 0x00000000004ec652 in PyRun_FileExFlags ()
#31 0x00000000004eae31 in PyRun_SimpleFileExFlags ()
#32 0x000000000049e14a in Py_Main ()
#33 0x00007ffff7811830 in __libc_start_main (main=0x49dab0 <main>, argc=2, 
    argv=0x7fffffffdc98, init=<optimized out>, fini=<optimized out>, 
    rtld_fini=<optimized out>, stack_end=0x7fffffffdc88)
    at ../csu/libc-start.c:291
#34 0x000000000049d9d9 in _start ()

```

Thanks

gortium, 2017-02-17:
+1 Ubuntu 16.04 kinetic
rqt or rosrun rqt_graph rqt_graph result in segmentation fault (core dumped)

dirk-thomas, 2017-02-17:
Please provide more information, e.g. the version numbers of all involved packages: Qt, `ros-kinetic-python-qt-binding`, `ros-kinetic-qt-gui`, `rqt`, PyQt / PySide, etc.

gortium, 2017-02-17:
ros-kinetic-python-qt-binding: 0.3.1-2xenial-20161026-181015-0700
ros-kinetic-qt-gui: 0.3.3-0xenial-20161026-190352-0700
ros-kinetic-rqt: 0.3.1-0xenial-20161115-173121-0800
python-pyqt5: 5.6+dfsg-1~cura~ppa~201609120727~6~ubuntu16.04.1
python3-pyqt5: 5.6+dfsg-1~cura~ppa~201609120727~6~ubuntu16.04.1
python-pyside2: 2.0.0+dev-0~201604151742~rev1858~pkg38~ubuntu16.04.1
 
That is what i got after installing ros-kinetic-descktop-full

dirk-thomas, 2017-02-17:
On Xenial Qt / PyQt version 5.5 is the default. Have you manually installed version 5.6? If yes, that is likely causing the segfault since the ROS binary packages are linked against version 5.5 and are likely not ABI compatible with a different minor releae.

gortium, 2017-02-17:
Synaptic say 5.6 is the lates version avalable.. When i try to force it to 5.5 it fail. 
I updated my kernel to the latest mainline to fix the not waking up issue. Could this be the cause ?

dirk-thomas, 2017-02-17:
According to the Ubuntu website Xenial contains 5.5: http://packages.ubuntu.com/search?suite=xenial&keywords=libqt5core5a

I don't know how you ended up with 5.6 but I am quite certain that is the problem. You can either try to get back to 5.5 or you have to rebuild all ROS Kinetic packages which depend on Qt (and use C++ code) from source.

gortium, 2017-02-17:
Ok thank you for your help! i will report back when it is done !!

dirk-thomas, 2017-02-17:
@juakofz I would be good if you could report back about your case too since it looks like this can be closed.

gortium, 2017-02-17:
OK i found why i had 5.6. Because i added a ppa needed to install cura. Cura probably need pyqt5.6. So after reinstalling pyqt5.5 everything work. Thank you !! 

I connot speak for @juakofz thought

dirk-thomas, 2017-02-17:
@gortium Thank you for checking and reporting back.

juakofz, 2017-02-17:
I eventually gave up and fixed my problem by reinstalled ubuntu, as I was using it solely for this purpose

closed 2017-02-17 by dirk-thomas

pirobot, 2019-09-25:
The only way I was able to fix this recently on my Ubuntu 16.04 machine (fully updated) was to build rqt from source using the kinetic-devel branch **and** add the following line to the end of my .bashrc file:

`export LD_LIBRARY_PATH=/opt/ros/kinetic/lib:$LD_LIBRARY_PATH`

If I try to use the ros-kinetic-rqt binary packages, I get the following error:

```
[rosrun] Couldn't find executable named rqt_gui below /opt/ros/kinetic/share/rqt_gui
[rosrun] Found the following, but they're either not files,
[rosrun] or not executable:
[rosrun]   /opt/ros/kinetic/share/rqt_gui
```

even though I can run other ROS kinetic nodes OK.

stevensgeek41, 2020-04-20:
I have the same issue, my qt version somehow ends up with qt5.11, I don't know how to reinstall it to qt5.5

dirk-thomas, 2020-04-20:
> I have the same issue, my qt version somehow ends up with qt5.11, I don't know how to reinstall it to qt5.5

Ubuntu Xenial ships with Qt 5.5 and that the version the Debian packages of ROS Kinetic are built against. Using a different Qt version at runtime is not going to work since different Qt minor versions are not ABI compatible.

stevensgeek41, 2020-04-21:
> > I have the same issue, my qt version somehow ends up with qt5.11, I don't know how to reinstall it to qt5.5
> 
> @stevensgeek41 Ubuntu Xenial ships with Qt 5.5 and that the version the Debian packages of ROS Kinetic are built against. Using a different Qt version at runtime is not going to work since different Qt minor versions are not ABI compatible.

Thanks for replying. I see. I think it's because I'm using KDE neon desktop. After I force to downgrade libqt5core5a, the KDE neon will not work. I end up in the recovery mode and reinstall the desktop. Anyway, for now I can tolerate the abscence of rqt.