ros-visualization/rqt #121: fatal error: rqt_gui_cpp/plugin.h: No such file or directory

Roysorus, 2017-06-15:
Hello,

I'm trying to build PTAM on ROS : http://wiki.ros.org/ethzasl_ptam but I'm getting the following error:
  `remote_ptam.h:38:32: fatal error: rqt_gui_cpp/plugin.h: No such file or directory`
I checked and rqt_gui_cpp/plugin.h does exist under /opt/ros/kinetic/include.
I'm working with Ros Kinetic on Ubuntu 16.04.
QMake version 2.01a
Using Qt version 4.8.7 in /usr/lib/x86_64-linux-gnu
Does anyone know what's wrong?

dirk-thomas, 2017-06-15:
Since the file exists (https://github.com/ros-visualization/rqt/blob/4eea6b68218af14fa21b962490c567260e06419a/rqt_gui_cpp/include/rqt_gui_cpp/plugin.h) you are likely not setting up the environment correctly. You might want to check your CMake logic if it sets the `include directories` correctly.

I am not sure if you have seen the [ROS support guidelines](http://wiki.ros.org/Support) but we kindly ask to raise questions like this on [answers.ros.org](http://answers.ros.org/) instead.

The rational for asking all kind of questions in a single place is:
* There are many more people reading questions there so your chances on getting an answer and in a timely manner are much higher.
* In the future other users will search there for similar problems and can find your question and the potential answers.
* The issue tracker in this repo is used to track bugs, feature requests, etc. But your question is not such an action item.

Therefore I will close this ticket here for now. If you have asked your question on answers.ros.org please feel free to add a link to the question to this ticket so that future readers can find the related question.

closed 2017-06-15 by dirk-thomas

pipigenius, 2018-10-20:
I meet the same problem, how did you solve it?
Thank you in advance.

pipigenius, 2018-10-20:
> Hello,
>
> I'm trying to build PTAM on ROS : http://wiki.ros.org/ethzasl_ptam but I'm getting the following error:
> `remote_ptam.h:38:32: fatal error: rqt_gui_cpp/plugin.h: No such file or directory`
> I [...]

Hi, I've met the same problem, could you plz tell me how did you solve it?

cumtchenchang, 2019-07-16:
Have you solved it?

icameling, 2019-12-28:
Hi

In addition to [ethzasl_ptam](http://wiki.ros.org/ethzasl_ptam ), what other open sources can be used to calibrate the camera with ATAN model?