ros-visualization/rqt #126: QXcbConnection

aravindharinair, 2017-12-12:
Hi.
Sorry for posting this as in issue in Github. I couldn't on ROS Answers as it's apparently down.
So, I'm learning ROS for a college project and I'm running Ubuntu 16.04 off Windows Subsystem for Linux as I'm not allowed to dual-boot my laptop. I don't have a spare machine lying around, so I'm running like this.
Now the problem is that when I run `rqt`, I'm getting an error stating:
`QXcbConnection: Could not connect to display :0`
On running `sudo apt-get install ros-lunar-rqt`, I get:
`Reading package lists... Done`
`Building dependency tree`
`Reading state information... Done`
`ros-lunar-rqt is already the newest version (0.5.0-0xenial-20171120-112204-0800).`
`0 upgraded, 0 newly installed, 0 to remove and 127 not upgraded.`
So, any help on how to get it running?
I'm running Xming as the X-Server.

dirk-thomas, 2017-12-12:
Please check back when answers

dirk-thomas, 2017-12-12:
Please check back when answers.ros.org is back online and then ask your question there. Please consider to post the link to the question here afterwards.

closed 2017-12-12 by dirk-thomas

aravindharinair, 2017-12-12:
Ok. Will do that once ROS Answers is online and get back to you.