ros-visualization/rqt #127: "PluginManager.__init__() multiprocess-mode only available under linux"

AndyZe, 2018-01-15:
I'm using Ubuntu 16.04 (Kinetic) and I believe X11 is running. `sudo Xorg -version` gives:
`X.Org X Server 1.19.5
Release Date: 2017-10-12
X Protocol Version 11, Revision 0`
`
top` shows Xorg running, and entering `sudo Xorg` from terminal gives:
`(EE) Server is already active for display 0`

phy25, 2018-03-28:
This seems like an issue in [qt_gui_core](https://github.com/ros-visualization/qt_gui_core).

dirk-thomas, 2018-04-02:
The multi-process mode is not supported in Qt 5. See ros-visualization/python_qt_binding#34 for more information why that is the case. The check was before only failing on non-Linux platforms since they didn't support the `X11Embed` API. In Qt 5 those have been removed entirely which makes the feature unsupported across all platforms and the error message kind of weird.

See ros-visualization/qt_gui_core#116 which removes the command line option.

closed 2018-04-02 by dirk-thomas