ros-visualization/rqt #130: rqt Segmentation fault since update

vitomillan, 2018-04-19:
I use ubuntu 16.04 and ros kinetics.
A few days ago rqt was updated by sudo apt-get update.

Since then I believe all rqt plugins/ applications fail due to 
`Segmentation fault`

I guess it might be due to python-pyqt4 vs python-pyqt5 dependencies.
I tried to remove python-pyqt4 and install python-pyqt5 to fix this.

`sudo apt-get install python-pyqt5`
`sudo apt-get remove python-pyqt4`

I guess this didn't work like I wanted. :(

This Script:

`from PyQt4.QtCore import QT_VERSION_STR
from PyQt4.pyqtconfig import Configuration
print("Qt version:", QT_VERSION_STR)
cfg = Configuration()
print("SIP version:", cfg.sip_version_str)
print("PyQt version:", cfg.pyqt_version_str)
`

(formatting didn't work here,sry!)
returns:

`('Qt version:', '4.8.7')
('SIP version:', '4.18')
('PyQt version:', '4.11.4')
`

dirk-thomas, 2018-04-19:
You seem to be using Qt 4 on your system which is not supported by `rqt` in `Kinetic`. This doesn't look like to be a problem in `rqt` itself.

If you need help debugging what is wrong on your system you might want to ask on answers.ros.org. I would recommend to try reinstalling the `rqt` packages (and its dependencies) again to ensure the necessary dependencies are being pulled in.

dirk-thomas, 2018-04-19:
It sounds like some Qt 4 specific package is interfering. I don't have any specific suggestion though. That is why I suggested answers.ros.org where more people will read your question and might have a suggestion.

vitomillan, 2018-04-19:
Then thanks anyway! I would like to wait for tommorrow if there's someone who has an idea on that. 
And I will go then for your advice of course!

vitomillan, 2018-04-20:
Hey Thomas! 

Thanks again! I moved this to answers.ros.org. Actually I still not have a clue about what to do, but maybe you will want to close this now...idk. xD

dirk-thomas, 2018-04-20:
> I moved this to answers.ros.org

It would be good to post the link to your question so that future readers can find it easily.

closed 2018-04-20 by dirk-thomas

vitomillan, 2018-04-25:
Sry..for late response. I wanted to add the link after question is solved and was not in town for some days. But rqt still isn't working. It's not pyqt4.

vitomillan, 2018-04-25:
https://answers.ros.org/question/253655/rqt-segmentation-fault-following-tutorials/

did the trick!