ros-visualization/rqt #131: rqt container plugin freeze

krixkrix, 2018-06-05:
I see frequent lockups when trying to place plugins inside the rqt _container_ widget.
For me it is extremely easy to reproduce:

1. launch roscore
2. launch an empty rqt window
3. launch a _container_ plugin from the rqt menu
4. launch any other plugin from the rqt menu, for example _Topics/Topic Monitor_
5. drag the _Topic Monitor_ widget over the _container_ widget and drop it
6. drag the _Topic Monitor_ widget away from the _container_ widget and drop it
7. repeat steps 5+6, after a few times the _Topic Monitor_ remains undocked and appears frozen.

It happens for any plugin, not just _Topic Monitor_

Apart from the problem being solved, I am also interesting to hear if anybody can reproduce this.

remod, 2019-02-08:
Me and some of my co-workers are experiencing the same issue. As soon as a container is part of the perspective, any plugin undocked and dragged freezes after before I am able to dock it again.

dirk-thomas, 2019-02-08:
@krixkrix @remod Please provide more information like platform, ROS distro, Qt version, which binding and version is being used etc.

mlautman, 2019-02-10:
I just reproduced the problem with the latest Kinetic Debian on a 16.04 machine

remod, 2019-02-11:
I am using 16.04/kinetic as well, with default installations of Qt (4.8 and 5.5).

remod, 2019-05-21:
For me, the issue persists on 18.04/melodic with Qt 5.9.5.

Mirrownis, 2022-02-10:
Could reproduce on 20.04/focal and galactic and found quick-fix: run 'rqt --clear-config' to remove all frozen plugins.

sloretz, 2025-06-02:
ROS 1 is end-of-life (EOL) as May 31st 2025, so I'm closing this.

[If you still rely on ROS 1, read this page to learn about your options.](https://bit.ly/NoeticEOL)

closed 2025-06-02 by sloretz