ros-visualization/rqt #132: No module named PySide2.QtCore

Redox15, 2018-06-06:


Redox15, 2018-06-06:
Can't run rqt, rqt_bag, rqt_image_view,...
Same error for all

> Traceback (most recent call last):
>   File "/opt/ros/melodic/bin/rqt", line 13, in <module>
>     sys.exit(main.main())
>   File "/opt/ros/melodic/lib/python2.7/dist-packages/rqt_gui/main.py", line 59, in main
>     return super(Main, self).main(argv, standalone=standalone, plugin_argument_provider=plugin_argument_provider, plugin_manager_settings_prefix=str(hash(os.environ['ROS_PACKAGE_PATH'])))
>   File "/opt/ros/melodic/lib/python2.7/dist-packages/qt_gui/main.py", line 340, in main
>     from python_qt_binding import QT_BINDING
>   File "/opt/ros/melodic/lib/python2.7/dist-packages/python_qt_binding/__init__.py", line 55, in <module>
>     from .binding_helper import loadUi, QT_BINDING, QT_BINDING_MODULES, QT_BINDING_VERSION  # @UnusedImport
>   File "/opt/ros/melodic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 252, in <module>
>     getattr(sys, 'SELECT_QT_BINDING_ORDER', None),
>   File "/opt/ros/melodic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 98, in _select_qt_binding
>     raise ImportError("Could not find Qt binding (looked for: %s):\n%s" % (', '.join(["'%s'" % b for b in binding_order]), '\n'.join(error_msgs)))
> ImportError: Could not find Qt binding (looked for: 'pyqt', 'pyside'):
>   ImportError for 'pyqt': /usr/lib/python2.7/dist-packages/PyQt5/QtCore.x86_64-linux-gnu.so: symbol _ZN23QOperatingSystemVersion11AndroidOreoE version Qt_5 not defined in file libQt5Core.so.5 with link time reference
> Traceback (most recent call last):
>   File "/opt/ros/melodic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 89, in _select_qt_binding
>     QT_BINDING_VERSION = binding_loader(required_modules, optional_modules)
>   File "/opt/ros/melodic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 131, in _load_pyqt
>     _named_import('PyQt5.%s' % module_name)
>   File "/opt/ros/melodic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 111, in _named_import
>     module = builtins.__import__(name)
> ImportError: /usr/lib/python2.7/dist-packages/PyQt5/QtCore.x86_64-linux-gnu.so: symbol _ZN23QOperatingSystemVersion11AndroidOreoE version Qt_5 not defined in file libQt5Core.so.5 with link time reference
> 
>   ImportError for 'pyside': No module named PySide2.QtCore
> Traceback (most recent call last):
>   File "/opt/ros/melodic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 89, in _select_qt_binding
>     QT_BINDING_VERSION = binding_loader(required_modules, optional_modules)
>   File "/opt/ros/melodic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 163, in _load_pyside
>     _named_import('PySide2.%s' % module_name)
>   File "/opt/ros/melodic/lib/python2.7/dist-packages/python_qt_binding/binding_helper.py", line 111, in _named_import
>     module = builtins.__import__(name)
> ImportError: No module named PySide2.QtCore

My LD_LIBRARY_PATH:
`/opt/Qt/5.9.1/gcc_64/lib:/home/andre/catkin_ws/devel/lib:/opt/ros/melodic/lib:/opt/Qt/5.9.1/gcc_64/lib:/opt/pleora/ebus_sdk/Ubuntu-x86_64/lib/genicam/bin/Linux64_x64:/opt/pleora/ebus_sdk/Ubuntu-x86_64/lib:/usr/local/lib`

A few weeks ago it was working. Any guess?

dirk-thomas, 2018-06-06:
What platform are you on? What Qt version do you have installed? What version of the ROS package `python_qt_binding` do you have installed?

Redox15, 2018-06-06:
Sorry, first time reporting an issue.

Ubuntu 18.04
Qt 5.9.1
How can I know the version of that package?

DorianScholz, 2018-06-06:
This looks like a version mismatch between your Qt version and your PyQt version.
Ubuntu 18.04 comes with Qt 5.9.5 and PyQt that was compiled against that specific version.
For some reason you have a local install of Qt 5.9.1 in your LD_LIBRARY_PATH which seems incompatible with the installed PyQt libraries.
Try removing the Qt 5.9.1 paths from LD_LIBRARY_PATH before running rqt.

Redox15, 2018-06-07:
Removing the Qt 5.9.1 paths from LD_LIBRARY_PATH kinda solve it because now I can run the rqt commands. However, using rqt_bag it wasn't possible to see the image on sensor_msgs/Image messages.

DorianScholz, 2018-06-07:
Sounds like the original issue is solved.
May I refer you to the ROS answers site for questions about using rqt_bag.

closed 2018-06-07 by DorianScholz

tobidelbruck, 2019-02-11:
I ran into same problem on virgin ubuntu 18.04 (but that had some tensorflow and nvidia driver stuff installed).  I finally solved it by first upgrading pip using the "python get-pip.py" method at https://pip.pypa.io/en/stable/installing/#upgrading-pip , and then using pip to install pyside2 with
 "pip install pyside2". 
You need to be outside any firewall for get-pip.py to run without knowing how to set the --proxy for get-pip.py to work inside a firewall.
Hope this helps someone else in same situation. This problem burned up several hours for this newbie.