ros-visualization/rqt #134: rqt fails on WSL

GavinKane, 2018-07-28:
On WSL with Ubuntu 16.04 xenial and Kernel: x86_64 Linux 4.4.0-17134-Microsoft
Using XMing on windows, and "export DISPLAY=:0" activated.
On executing **rqt**, it fails and gives the error:
```
failed to get the current screen resources
The X11 connection broke: I/O error (code 1)
XIO:  fatal IO error 2 (No such file or directory) on X server ":0"
      after 349 requests (349 known processed) with 0 events remaining.
```

dirk-thomas, 2018-07-31:
WSL is not a platform we usually test on / support. Please consider to provide a pull request for supporting this.

GavinKane, 2018-07-31:
I can however it confirms with VcXsrv.
Thanks to eric-wieser for the hint, who said 
> This might just be a bug in XMing, for which the free version is stuck at an old version of X11

stewpend0us, 2022-10-15:
Yeah this works with VcXsrv for me