ros-visualization/rqt #135: rqt_plot cannot receive any data

Herschelpu, 2018-08-01:
I tried to plot my data into rqt_plot to get the figure. However rqt suddenly return zeros for all of my data. I checked the data by using "rostopic echo" command and it actually has non-zero data stream. I created a node that only had one publisher publishing a non-zero number but also got zero plot. I also tried to reinstall this ros-kinetic-rqt but still not working..... so is there anyone know what happen to my rqt plot?     
[image]

dirk-thomas, 2018-08-03:
I tried to publish a number using the command line tools:

```
rostopic pub /number std_msgs/Int32 "{data: 42}" -r 10
```

And after selecting the topic `/number` in the `rqt_plot` gui the line is plotted at the expected value.

dirk-thomas, 2019-07-29:
Closing due to no responses.

closed 2019-07-29 by dirk-thomas