ros-visualization/rqt #136: Consolidating RQT packages

mlautman, 2018-10-11:
Hello maintainers!

PickNik is working on a port of RQT to ROS2. We would like to make these changes for ROS2 in a fork so that we can cherry-pick commits back to the upstream repo. In order to do this we will have to consolodate the RQT packages into a single repository that we can fork to the ROS2 organization. 

The MoveIt! project did the same thing a few years ago and it has made it much easier for maintainers and contributors to make changes to MoveIt!. For this effort we would like to take the same approach as is documented [here](https://github.com/davetcoleman/moveit_merge)

Are there any strong arguements for keeping the ros-visualization RQT packages separate?

dirk-thomas, 2018-10-11:
Are you referring to the repositories containing the rqt plugins?

dirk-thomas, 2018-10-11:
> PickNik is working on a port of RQT to ROS2.

That is great. Please share this also on discourse to make more people aware of this (to avoid duplicate efforts).

> We would like to make these changes for ROS2 in a fork so that we can cherry-pick commits back to the upstream repo.

In general we don't recommend forking existing 
I encourage you to (at least long term) not use forked repos but to contribute a `ros2` branch in the existing repositories. For initial development having a forked repo should be fine. If you want me to create `ros2` branches so that you can target them with PRs please let me know (or create a ticket requesting such branch on the repository you need it on e.g. where I am not a maintainer).

> In order to do this we will have to consolidate the RQT packages into a single repository that we can fork to the ROS2 organization.

You can achieve the same with separate repositories - so re-merging them is not a requirement for this porting effort.

> Are there any strong arguements for keeping the ros-visualization RQT packages separate?

These repositories have intentionally been separated from the rqt framework repository a while ago and there is no intention to revert this. The advantages to keep them separate outweigh the convenience of a single monolithic repository:
* separate maintainers are in charge of issues and pull requests of a specific plugin (or set of packages)
* separate maintainers can release their packages at their own timeline and pace when required
* a patch release for one plugin doesn't force an unnecessary release on all the other packages in the repo

mlautman, 2018-10-11:
@dirk-thomas Thanks for the quick reply! 

I actually just posted on Discourse earlier today. I have added the link above :) 

Let's discuss offline and come back to this once we've come to a resolution on the best strategy moving forward.

closed 2018-10-11 by mlautman