ros-visualization/rqt #165: ROS2 port: Button icons don't show in Mac/Windows

brawner, 2018-12-04:
Buttons for several plugins don't show the icons that show up in linux. For example, the add/remove/clear publisher(s) in rqt_publisher.

wjwwood, 2018-12-14:
Also, the buttons which are common to all "panels" (not sure that's the right terminology) are very small and as a result have terrible aliasing:

[image]

Definitely only a cosmetic thing though.

brawner, 2018-12-14:
@wjwwood Did you install the icons through homebrew?

brawner, 2018-12-14:
I created a new package [qt_gui_icons](https://github.com/ros-visualization/qt_gui_icons) that pulls in the scalable Tango icons. I created a PR in qt_gui_core to search for the icons (https://github.com/ros-visualization/qt_gui_core/pull/153).

That might give you nicer icons, but on a mac it requires building with the cmake flag -DINSTALL_QT_GUI_ICONS=TRUE

windelbouwman, 2019-06-17:
I can confirm this issue on Windows 10. I installed the dashing diademata via the zip file method. Most icons are not visible in rqt.

ahcorde, 2020-05-29:
This issue is still open for Foxy. Proposed solution

 - On windows
    - Add https://github.com/ros-visualization/qt_gui_icons to ros2.repos to fix the icons
    - Add the logic to load the icons when there is no theme available
 - On Mac there are two options:
   - As discussed [here](https://github.com/ros-visualization/rqt/issues/165#issuecomment-447483546). There is a formula to install icons on MacOS. No changes in the code are required
  - Modify qt_gui_icons to install the icons also on MacOS

@wjwwood @dirk-thomas what is the prefered solution for MacOS?

wjwwood, 2020-05-29:
Having it install better ones on macOS too is fine by me. Or we could try to update the Homebrew formula, I actually don't mind which.

Hello970, 2020-08-21:
Hi, I met the same problem and tried the method proposed above. However, the problem still has not been solved. If possible, would you mind giving a specific introduction about how to use the above package (https://github.com/ros-visualization/tango_icons_vendor) solving the problem of icons? Thank you very much. @ahcorde

dirk-thomas, 2020-09-23:
With ros-visualization/qt_gui_core#222 released the problem should be solved. Assuming that is the case I will go ahead and close this ticket.

If the problem persists please check that you are using the latest version, share enough context about the platform and the versions of involved packages you use and the ticket can be reopened if needed.

closed 2020-09-23 by dirk-thomas

Aposhian, 2021-04-12:
I am getting this same issue, but with docker.

Host OS: Ubuntu 18.04 LTS
RQT Version: 1.07

Dockerfile
```Dockerfile
FROM ros:foxy-ros-core
RUN apt-get update && apt-get install -y ros-${ROS_DISTRO}-rqt-plot
CMD ros2 run rqt_plot rqt_plot  
```

Command
```
docker run -it -e DISPLAY -e TERM -e QT_X11_NO_MITSHM=1 -e XAUTHORITY=/tmp/.docker.xauth -v /tmp/.docker/.xauth:/tmp/.docker.xauth -v /tmp/.X11-unix:/tmp/.X11-unix -v /etc/localtime:/etc/localtime:ro
```

Result:
[image]

clalancette, 2021-04-13:
#249 is tracking this issue for Galactic; once we find the problem, we may be able to backport it to Foxy.

ohh-haolin, 2022-01-01:
I find an interesting answer in Linux and Mac -- an undefined or unexpected behaviour caused by security policies in the desktop environment. If we log into a DE as user A and launch a qt-based program in user B by su/sudo/others, some external assets, like icons, cannot be properly loaded. This could explain situations within docker and other similar problems I met.