ros-visualization/rqt #181: ros2: Destroying subscription crashes python rqt_gui

brawner, 2018-12-13:
There is something not thread about creating a subscription in a rclpy plugin and then destroying it later in the same plugin. This was the stacktrace when python crashed.

```
Thread 7 Crashed:: RclpySpinner
0   librmw_fastrtps_shared_cpp.dylib	0x000000011c06747e SubListener::hasData() + 126
1   librmw_fastrtps_shared_cpp.dylib	0x000000011c067213 check_wait_set_for_data(rmw_subscriptions_t const*, rmw_guard_conditions_t const*, rmw_services_t const*, rmw_clients_t const*) + 115
2   librmw_fastrtps_shared_cpp.dylib	0x000000011c068a97 rmw_fastrtps_shared_cpp::__rmw_wait(rmw_subscriptions_t*, rmw_guard_conditions_t*, rmw_services_t*, rmw_clients_t*, rmw_wait_set_t*, rmw_time_t const*)::$_0::operator()() const + 39
3   librmw_fastrtps_shared_cpp.dylib	0x000000011c068b0f bool std::__1::condition_variable::wait_until<std::__1::chrono::steady_clock, std::__1::chrono::duration<long long, std::__1::ratio<1l, 1000000000l> >, rmw_fastrtps_shared_cpp::__rmw_wait(rmw_subscriptions_t*, rmw_guard_conditions_t*, rmw_services_t*, rmw_clients_t*, rmw_wait_set_t*, rmw_time_t const*)::$_0>(std::__1::unique_lock<std::__1::mutex>&, std::__1::chrono::time_point<std::__1::chrono::steady_clock, std::__1::chrono::duration<long long, std::__1::ratio<1l, 1000000000l> > > const&, rmw_fastrtps_shared_cpp::__rmw_wait(rmw_subscriptions_t*, rmw_guard_conditions_t*, rmw_services_t*, rmw_clients_t*, rmw_wait_set_t*, rmw_time_t const*)::$_0) + 95
4   librmw_fastrtps_shared_cpp.dylib	0x000000011c068051 rmw_fastrtps_shared_cpp::__rmw_wait(rmw_subscriptions_t*, rmw_guard_conditions_t*, rmw_services_t*, rmw_clients_t*, rmw_wait_set_t*, rmw_time_t const*) + 2561
5   librmw_fastrtps_cpp.dylib     	0x000000011c005dcd rmw_wait + 61
6   librmw_implementation.dylib   	0x00000001097a4fa4 rmw_wait + 132
7   librcl.dylib                  	0x00000001096ecf71 rcl_wait + 2465
8   _rclpy.cpython-37m-darwin.so  	0x00000001096be838 rclpy_wait + 200
9   org.python.python             	0x0000000108bc420f _PyMethodDef_RawFastCallKeywords + 236
10  org.python.python             	0x0000000108bc38af _PyCFunction_FastCallKeywords + 44
11  org.python.python             	0x0000000108c59b2b call_function + 636
12  org.python.python             	0x0000000108c52771 _PyEval_EvalFrameDefault + 7016
13  org.python.python             	0x0000000108bcf4bf gen_send_ex + 242
14  org.python.python             	0x0000000108c4f111 builtin_next + 99
15  org.python.python             	0x0000000108bc4313 _PyMethodDef_RawFastCallKeywords + 496
16  org.python.python             	0x0000000108bc38af _PyCFunction_FastCallKeywords + 44
17  org.python.python             	0x0000000108c59b2b call_function + 636
18  org.python.python             	0x0000000108c5280f _PyEval_EvalFrameDefault + 7174
19  org.python.python             	0x0000000108c5a432 _PyEval_EvalCodeWithName + 1835
20  org.python.python             	0x0000000108bc3874 _PyFunction_FastCallKeywords + 225
21  org.python.python             	0x0000000108c59ba0 call_function + 753
22  org.python.python             	0x0000000108c528b5 _PyEval_EvalFrameDefault + 7340
23  org.python.python             	0x0000000108c5a432 _PyEval_EvalCodeWithName + 1835
24  org.python.python             	0x0000000108bc3874 _PyFunction_FastCallKeywords + 225
25  org.python.python             	0x0000000108c59ba0 call_function + 753
26  org.python.python             	0x0000000108c528b5 _PyEval_EvalFrameDefault + 7340
27  org.python.python             	0x0000000108bc3c8a function_code_fastcall + 112
28  org.python.python             	0x0000000108bc460d _PyObject_Call_Prepend + 150
29  org.python.python             	0x0000000108bc39bd PyObject_Call + 136
30  sip.so                        	0x000000010a58a122 call_method + 76
31  sip.so                        	0x000000010a585f78 sip_api_call_procedure_method + 152
32  QtCore.so                     	0x0000000109de42e4 sipQThread::run() + 84
33  org.qt-project.QtCore         	0x0000000109fea75e 0x109fbc000 + 190302
34  libsystem_pthread.dylib       	0x00007fff5af87661 _pthread_body + 340
35  libsystem_pthread.dylib       	0x00007fff5af8750d _pthread_start + 377
36  libsystem_pthread.dylib       	0x00007fff5af86bf9 thread_start + 13
```

dirk-thomas, 2018-12-13:
Almost all rmw functions are marked as non-thread-safe. Therefore I don't think you can expect it to work when calling functions from one thread while waiting / spinning in another QThread.

brawner, 2018-12-13:
@dirk-thomas Ok, then this package needs to handle that somehow. Why mark it invalid?

dirk-thomas, 2019-07-29:
This has been resolved with ros2/rclpy#318.

closed 2019-07-29 by dirk-thomas