ros-visualization/rqt #183: ros2: cannot click the menus on macOS

wjwwood, 2018-12-14:
Testing a release candidate of ROS 2 Crystal on macOS, and I just run rqt, but after it opens I cannot click the menu bar (never seen this before in other applications on my mac). If I wait a long time it still is not possible to clicking. However, if I change applications and then change back again I can access the menu bars 🙃.

Here's a gif:

[image]

wjwwood, 2018-12-15:
No that doesn't make a difference. If I open it from a not full-screen terminal, then it still doesn't let me click the menu's right off the bat.

pdalloz, 2019-05-30:
This happens when PyQt apps are called from python.exe rather than pythonw.exe
python.exe is for running command line scripts, and pythonw.exe is for running GUIs

Use pythonw.exe

omerts, 2022-02-17:
> This happens when PyQt apps are called from python.exe rather than pythonw.exe python.exe is for running command line scripts, and pythonw.exe is for running GUIs
> 
> Use pythonw.exe

@pdalloz This is mac bro