ros-visualization/rqt #187: [ROS2] QOS profile options and subscribing problem

Myzhar, 2019-01-17:
I'm testing RQT on ROS2 Crystal Clemmys and I noticed that I cannot subscribe image topics using the `rqt_image_view` plugin.

The problem is due to QOS settings since I'm not publishing the Image topics using the default values:
https://index.ros.org/doc/ros2/Concepts/About-Quality-of-Service-Settings/#qos-compatibilities

I think that each GUI plugin should provide a way to set the QoS setting to be used for each topic that can be subscribed.

anaelle-sw, 2020-05-20:
Hi, I encounter the same kind of problem here.

System:
ROS2 Foxy, and running the demo world from gazebo_ros_pkgs ` gazebo_ros_ray_sensor_demo.world `. 
(I have also noticed the same issue showing up on Eloquent and Dashing, either with this demo world or with custom sensors.)

Description:
The topics for sensors' feedback (laserscan, range, etc) are available on rqt, and I can suscribe to them via the topic monitor pannel. But no message is received by rqt, while messages are actually published on the topics. This can be checked either with ` ros2 topic echo ` command line or on rviz2. 
I think this is due to QOS profiles compatibility since, while using rviz2, the topics profiles have to be changed from reliable to best effort in order to visualize the sensors' feedback. So I agree it would be nice to be able to change QOS profiles from rqt GUI.

good0613, 2020-07-25:
Hi, I also have same issue.
System:
- ROS2 foxy and with gazebo camera demo world;
>gazebo --verbose /opt/ros/foxy/share/gazebo_plugins/worlds/gazebo_ros_camera_demo.world &
>rviz2&

Add topic `/demo_cam/camera1/image_raw` to `rviz2` --> no image is displayed.

Description:
- topic is being published with following command;
>ros2 topic hz /demo_cam/camera1/image_raw

As @anaelle-sw mentioned, it's same on `gazebo_ros_ray_sensor_demo.world`; No scan data on `rviz2`...

Thanks.

LouisChen1905, 2020-08-24:
Any idea on how to fix it? @dirk-thomas

dirk-thomas, 2020-08-26:
Imo the options are:

* introspecting the graph and use the QoS matching existing endpoints
* a configuration option to choose the QoS setting per plugin (if a plugin uses multiple topics potentially for each topic separately)

jacobperron, 2021-02-02:
FWIW, as of ROS Galactic, there exists a feature to enable configuration QoS with ROS parameters:
https://index.ros.org/doc/ros2/Releases/Release-Galactic-Geochelone/#externally-configure-qos-at-start-up

I suggest we enable this feature for rqt plugins.

808brick, 2021-03-30:
Just as an FYI if people are confused, when they are talking about changing the QoS policies in RViz, they mean through the options in the side panel here:

[image]

808brick, 2021-03-30:
I know Quality of Service can be confusing to some people, so in case you run into an issue of not being able to have your nodes subscribe to Gazebo plugin publishers, you can use the `ros2 topic info /topic_name --verbose` command to view the QoS  configuration of the Publishers and Subscribers of that topic. 

To configure QoS properly in Python, for example in a subscriber, you can use the following:
```
qos_policy = rclpy.qos.QoSProfile(reliability=rclpy.qos.ReliabilityPolicy.BEST_EFFORT,
                                          history=rclpy.qos.HistoryPolicy.KEEP_LAST,
                                          depth=1)
self.sub = self.create_subscription(Image, topic,
                                    self.subscriber_callback, qos_profile=qos_policy)
```

The above is a subscriber I made to retrieve images from my gazebo simulation using the `libgazebo_ros_camera.so` plugin.

Hopefully this helps someone else down the line.

andreasBihlmaier, 2021-12-26:
This issue still persists today, e.g. with galactic.

When trying to use the most straightforward combination of a Gazebo camera plugin (libgazebo_ros_camera.so) and rqt image view, one will not see the camera images, rather one will get the following message from rqt:
```
[WARN] [...] [rqt_gui_cpp_node_2726]: New publisher discovered on topic '/camera/image_raw', offering incompatible QoS. No messages will be sent to it. Last incompatible policy: RELIABILITY_QOS_POLICY
```

I'd consider that there is something broken in the ROS2 ecosystem, if this simple use case doesn't work.
The same goes for rviz. Although it now has QoS parameters (Depth, History, Reliability, Durability) exposed in the UI, it feels wrong for users having to fiddle with them when using default settings everywhere.

Likely two things are required to fix this:
- Agree on the default policy for sensor_msgs/Image and use this everywhere (Gazebo plugins, rviz, rqt)
- Expose QoS parameters everywhere (in UI, in configuration file or through environment parameters)

karl-schulz, 2022-01-28:
Would it be an option to set the default QoS for subscribers in rqt_image_view to BEST_EFFORT? Most image data is published via BEST_EFFORT anyways, and messages with RELIABLE will still be shown (it's compatible in that direction).

Obviously, having an option like in RVIZ would be ideal, but probably also more work.

jacobperron, 2022-02-15:
FWIW, I think you can reconfigure all of the default gazebo_ros plugins via XML (in the URDF/SDF) since Foxy. Here is a link to some documentation:
https://github.com/ros-simulation/gazebo_ros_pkgs/blob/205755685f77300a27d16c9739b1e397e30811d9/gazebo_ros/include/gazebo_ros/qos.hpp#L61-L120

---

Another option is to dynamically select the QoS based on the publisher. Here is a related issue to add a common API to support such a feature in ROS 2: https://github.com/ros2/rclcpp/issues/1868

m2-farzan, 2022-03-15:
Thanks for the hint, @jacobperron

I added the `<qos>` tag to the `<plugin>` tag of my camera like this:

```xml
<plugin name="camera_plugin" filename="libgazebo_ros_camera.so">
    <ros>
        <namespace>camera</namespace>
        <qos>
            <topic name="/camera/${side}/image_raw">
                <publisher>
                    <reliability>reliable</reliability>
                </publisher>
            </topic>
        </qos>
    </ros>
    <camera_name>${side}</camera_name>
    <frame_name>camera_${side}_link_optical</frame_name>
    <hack_baseline>0.2</hack_baseline>
</plugin>
```

It solved the qos compatibility problem with rqt.

rosiakpiotr, 2023-12-29:
This issue was opened on beginning of 2019, we have almost 2024 and still no resolution? Any updates?

karl-schulz, 2024-04-09:
> This issue was opened on beginning of 2019, we have almost 2024 and still no resolution? Any updates?

Pretty sure it has been fixed by images being subscribed as BEST_EFFORT now.
See https://github.com/ros-visualization/rqt_image_view/pull/67

wojtess, 2025-07-04:
Is there any update on that? I am using rtabmap and /odom topic is published with BEST_EFFORT. I cant find anywhere option to change QoS in rqt topic monitor.

rosiakpiotr, 2025-07-12:
> Is there any update on that? I am using rtabmap and /odom topic is published with BEST_EFFORT. I cant find anywhere option to change QoS in rqt topic monitor.

Please check [this answer](https://github.com/ros-visualization/rqt_topic/issues/49#issuecomment-3066216042) - it describes case with `BEST_EFFORT` for topic monitor.