ros-visualization/rqt #192: [crystal-devel] rqt_py_common test failing

jacobperron, 2019-03-30:
Linux CI: [image]

```
build/rqt_py_common/test_results/rqt_py_common/rqt_py_common.xunit.xml: 12 tests, 0 errors, 1 failure, 0 skipped
- rqt_py_common.test.test_rqt_common_unit.TestMessageTreeModel test_path_names (test/test_rqt_common_unit.py:39)
  <<< failure message
    AttributeError: 'NoneType' object has no attribute '_path'
  >>>
```

jacobperron, 2019-04-05:
@dirk-thomas I think this can be close?

dirk-thomas, 2019-04-05:
Interesting, the reference in the description of the merged PR should have done that automatically...

Thanks for pointing it out - closing it manually.

PR #193 (2019-04-05): accept array.array and numpy.ndarray
Fixes #192.

[image]

While not necessary for Crystal but only Dashing it doesn't harm being on the `crystal` branch and avoids creating a `dashing` branch.
closed 2019-04-05 by dirk-thomas