ros-visualization/rqt #195: rqt on ROS2 - cannot export perspective - 'QtCore' is not defined

bergercookie, 2019-05-15:
I've launched rqt and I'm getting the following error when I'm trying to save the current layout to a perspective file:

```
Traceback (most recent call last):
  File "/opt/ros/crystal/lib/python3.6/site-packages/qt_gui/perspective_manager.py", line 188, in save_settings_completed
    callback(*callback_args)
  File "/opt/ros/crystal/lib/python3.6/site-packages/qt_gui/perspective_manager.py", line 399, in _on_export_perspective_continued
    self._convert_values(data, self._export_value)
  File "/opt/ros/crystal/lib/python3.6/site-packages/qt_gui/perspective_manager.py", line 423, in _convert_values
    self._convert_values(groups[group], convert_function)
  File "/opt/ros/crystal/lib/python3.6/site-packages/qt_gui/perspective_manager.py", line 420, in _convert_values
    keys[key] = convert_function(keys[key])
  File "/opt/ros/crystal/lib/python3.6/site-packages/qt_gui/perspective_manager.py", line 460, in _export_value
    reimported = self._import_value(data)
  File "/opt/ros/crystal/lib/python3.6/site-packages/qt_gui/perspective_manager.py", line 429, in _import_value
    return QByteArray.fromHex(eval(value['repr(QByteArray.hex)']))
  File "<string>", line 1, in <module>
NameError: name 'QtCore' is not defined
```

P.S. I've opened a question on ROS answers about this but most probably this is a bug, so I'm also reporting here, in case you want to address it at some point (https://answers.ros.org/question/322656/rqt-on-ros2-cannot-export-perspective-qtcore-is-not-defined/)

dirk-thomas, 2019-05-15:
For the trivial case with no plugins open creating a new perspective works just fine. Please include enough information to reproduce the problem.

bergercookie, 2019-05-16:
Here's the percpective file I'm using:

[demo.perspective.txt](https://github.com/ros-visualization/rqt/files/3186722/demo.perspective.txt)

I'm working on Ubuntu 18.04 using the vanilla `crystal` distribution from ROS Debian repositories and have installed all `rqt-*` debian packages.

Let me know if you need any additional information and whether you can reproduce it.

dirk-thomas, 2019-06-10:
Fixed by ros-visualization/qt_gui_core#178.

closed 2019-06-10 by dirk-thomas