ros-visualization/rqt #197: Malformed message_type rqt ROS2

skucheria, 2019-05-29:
**- Operating System:**
Ubuntu 18.04
**- Installation type:**
from source
**- Version or commit hash:**
current master
**- DDS implementation:**
RTI Connext
**- Client library (if applicable):**
rqt

Receiving "malformed message_type" error on message types in rqt when attempting to use rqt_plot or topic monitor. 
Ex:

```
[ERROR] [get_message_class]: Malformed message_type: std_msgs/msg/String
TopicInfo.__init__(): topic "/robot_description": can not get message class for type "std_msgs/msg/String"
```

dirk-thomas, 2019-05-29:
This should be resolved on the latest HEAD on the `crystal-devel` branch. PlLease try it again to confirm that it is fixed for you.

dirk-thomas, 2019-05-29:
I will go ahead and close this assuming the problem has already been resolved. Please feel free to comment on the closed ticket and if necessary it can be reopened.

closed 2019-05-29 by dirk-thomas