ros-visualization/rqt #198: [ROS2] Traceback when closing/exiting

Karsten1987, 2019-05-29:
When using `rqt_publisher` under dashing, I get a traceback when closing the application on OSX. This might be related to breaking changes in dashing related to passing the context object to `rcl_init()`. 
See here: https://github.com/ros2/rcl/blob/master/rcl/include/rcl/init.h#L82 and the related PR: https://github.com/ros2/rcl/pull/336

```
Traceback (most recent call last):
  File "/Users/karsten/workspace/osrf/ros2_full/build/rqt_gui_py/src/rqt_gui_py/rclpy_spinner.py", line 32, in run
    executor.spin_once(timeout_sec=1.0)
  File "/Users/karsten/workspace/osrf/ros2_full/install/lib/python3.7/site-packages/rclpy/executors.py", line 695, in spin_once
    handler, entity, node = self.wait_for_ready_callbacks(timeout_sec=timeout_sec)
  File "/Users/karsten/workspace/osrf/ros2_full/install/lib/python3.7/site-packages/rclpy/executors.py", line 649, in wait_for_ready_callbacks
    return next(self._cb_iter)
  File "/Users/karsten/workspace/osrf/ros2_full/install/lib/python3.7/site-packages/rclpy/executors.py", line 532, in _wait_for_ready_callbacks
    self._context.handle)
RuntimeError: Failed to initialize wait set: the given context is not valid, either rcl_init() was not called or rcl_shutdown() was called., at /Users/karsten/workspace/osrf/ros2_full/src/ros2/rcl/rcl/src/rcl/wait.c:130
```

dirk-thomas, 2019-05-29:
Does this only happen on macOS? Can you please try it on Ubuntu too. I can't reproduce the problem by just starting it with `ros2 run rqt_publisher rqt_publisher` and closing it through the [X] button.

Karsten1987, 2019-05-29:
I currently don't have access to a graphical Ubuntu.
It could definitely just be a OSX problem - yet I can reproduce it in 100% of the cases.

Karsten1987, 2019-05-29:
So I've tried it on a 18.04 docker instance and see the same error message when closing the app.

dirk-thomas, 2019-05-29:
Similar to ros2/demos#316.

dirk-thomas, 2020-06-12:
@Karsten1987 This seems to be a problem outside of `rqt`, I tried to reproduce the problem on Foxy but the error doesn't show up. Can you check if the problem is still happening for you in the latest distro Foxy.

hdino, 2020-06-13:
@dirk-thomas For me both this issue and #207 no longer show up on Foxy (Ubuntu 20.04).

dirk-thomas, 2020-06-13:
@hdino Thanks for checking it. I will go ahead and close the ticket then.

closed 2020-06-13 by dirk-thomas