ros-visualization/rqt #204: Ubuntu 14.04, ROS Indigo: Failure to start RViz plugin, "Method "start_plugin" ... doesn't exist"

harel55, 2019-07-09:
Attempting to run `rqt --command-start-plugin rqt_rviz` results in the terminal output

`Traceback (most recent call last):
  File "/opt/ros/indigo/bin/rqt", line 13, in <module>
    sys.exit(main.main())
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rqt_gui/main.py", line 59, in main
    return super(Main, self).main(argv, standalone=standalone, plugin_argument_provider=plugin_argument_provider, plugin_manager_settings_prefix=str(hash(os.environ['ROS_PACKAGE_PATH'])))
  File "/opt/ros/indigo/lib/python2.7/dist-packages/qt_gui/main.py", line 324, in main
    (rc, msg) = remote_interface.start_plugin(self._options.command_start_plugin, ' '.join(self._options.plugin_args))
  File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 70, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib/python2.7/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python2.7/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownMethod: Method "start_plugin" with signature "ss" on interface "org.ros.qt_gui.PluginManager" doesn't exist`

despite the fact that RViz appears in the output of `rqt --list-plugins`. I was unable to find any information on the particular exception this caused.

dirk-thomas, 2019-07-24:
ROS Indigo has been EOL-ed already (see https://discourse.ros.org/t/confusion-about-eol-for-indigo/9065/2).

Please try to reproduce the problem with an active ROS distribution (currently Kinetic and Melodic) and comment again.

dirk-thomas, 2019-07-29:
I am closing this ticket for now due to missing information. Please feel free to comment on the closed ticket with the requested information.

closed 2019-07-29 by dirk-thomas