ros-visualization/rqt #207: [ROS2] Traceback/exception when closing/exiting (Qt DockWidget)

hdino, 2019-08-19:
I've installed dashing on a fresh Ubuntu 18.04 and encounter the following errors when closing rqt (no matter if it's closed using the menu, Ctrl+C or the window's X):

```
Traceback (most recent call last):
  File "/opt/ros/dashing/lib/python3.6/site-packages/rqt_gui_py/rclpy_spinner.py", line 32, in run
    executor.spin_once(timeout_sec=1.0)
  File "/opt/ros/dashing/lib/python3.6/site-packages/rclpy/executors.py", line 695, in spin_once
    handler, entity, node = self.wait_for_ready_callbacks(timeout_sec=timeout_sec)
  File "/opt/ros/dashing/lib/python3.6/site-packages/rclpy/executors.py", line 649, in wait_for_ready_callbacks
    return next(self._cb_iter)
  File "/opt/ros/dashing/lib/python3.6/site-packages/rclpy/executors.py", line 532, in _wait_for_ready_callbacks
    self._context.handle)
RuntimeError: Failed to initialize wait set: the given context is not valid, either rcl_init() was not called or rcl_shutdown() was called., at /tmp/binarydeb/ros-dashing-rcl-0.7.6/src/rcl/wait.c:130
Exception ignored in: <bound method DockWidgetTitleBar.__del__ of <qt_gui.dock_widget_title_bar.DockWidgetTitleBar object at 0x7f3b6e340828>>
Traceback (most recent call last):
  File "/opt/ros/dashing/lib/python3.6/site-packages/qt_gui/dock_widget_title_bar.py", line 109, in __del__
    self._dock_widget.removeEventFilter(self)
RuntimeError: wrapped C/C++ object of type DockWidget has been deleted
```

or

```
Exception ignored in: <bound method DockWidgetTitleBar.__del__ of <qt_gui.dock_widget_title_bar.DockWidgetTitleBar object at 0x7f4c84f37ca8>>
Traceback (most recent call last):
  File "/opt/ros/dashing/lib/python3.6/site-packages/qt_gui/dock_widget_title_bar.py", line 109, in __del__
    self._dock_widget.removeEventFilter(self)
RuntimeError: wrapped C/C++ object of type DockWidget has been deleted
Exception ignored in: <bound method DockWidgetTitleBar.__del__ of <qt_gui.dock_widget_title_bar.DockWidgetTitleBar object at 0x7f4c84df8438>>
Traceback (most recent call last):
  File "/opt/ros/dashing/lib/python3.6/site-packages/qt_gui/dock_widget_title_bar.py", line 109, in __del__
    self._dock_widget.removeEventFilter(self)
RuntimeError: wrapped C/C++ object of type DockWidget has been deleted
```

or

```
Exception ignored in: <bound method DockWidgetTitleBar.__del__ of <qt_gui.dock_widget_title_bar.DockWidgetTitleBar object at 0x7feb2a34eca8>>
Traceback (most recent call last):
  File "/opt/ros/dashing/lib/python3.6/site-packages/qt_gui/dock_widget_title_bar.py", line 109, in __del__
    self._dock_widget.removeEventFilter(self)
RuntimeError: wrapped C/C++ object of type DockWidget has been deleted
```

or (Ctrl+C)

```
^C[INFO] [rclcpp]: signal_handler(signal_value=2)
Traceback (most recent call last):
  File "/opt/ros/dashing/lib/python3.6/site-packages/rqt_gui_py/rclpy_spinner.py", line 32, in run
    executor.spin_once(timeout_sec=1.0)
  File "/opt/ros/dashing/lib/python3.6/site-packages/rclpy/executors.py", line 695, in spin_once
    handler, entity, node = self.wait_for_ready_callbacks(timeout_sec=timeout_sec)
  File "/opt/ros/dashing/lib/python3.6/site-packages/rclpy/executors.py", line 649, in wait_for_ready_callbacks
    return next(self._cb_iter)
  File "/opt/ros/dashing/lib/python3.6/site-packages/rclpy/executors.py", line 532, in _wait_for_ready_callbacks
    self._context.handle)
RuntimeError: Failed to initialize wait set: the given context is not valid, either rcl_init() was not called or rcl_shutdown() was called., at /tmp/binarydeb/ros-dashing-rcl-0.7.6/src/rcl/wait.c:130
Exception ignored in: <bound method DockWidgetTitleBar.__del__ of <qt_gui.dock_widget_title_bar.DockWidgetTitleBar object at 0x7fe9d403aca8>>
Traceback (most recent call last):
  File "/opt/ros/dashing/lib/python3.6/site-packages/qt_gui/dock_widget_title_bar.py", line 109, in __del__
    self._dock_widget.removeEventFilter(self)
RuntimeError: wrapped C/C++ object of type DockWidget has been deleted
```

Karsten1987, 2019-08-19:
I think that's related to https://github.com/ros-visualization/rqt/issues/198

dirk-thomas, 2020-06-12:
The first error:

> RuntimeError: Failed to initialize wait set: the given context is not valid, either rcl_init() was not called or rcl_shutdown() was called., at /tmp/binarydeb/ros-dashing-rcl-0.7.6/src/rcl/wait.c:130

seems to be a problem outside of `rqt`. I tried to reproduce the problem on Foxy but the error doesn't show up. Can you check if the problem is still happening for you in the latest distro Foxy.

---

The second error:

> RuntimeError: wrapped C/C++ object of type DockWidget has been deleted

has been resolved by ros-visualization/qt_gui_core#213. I just made releases with this fix for Dashing and Eloquent.

dirk-thomas, 2020-06-12:
Since the first issue is tracked in #198 I will close this as a duplicate.

closed 2020-06-12 by dirk-thomas