ros-visualization/rqt #213: Inconsistent ordering of sensor_msgs/PointField

pauljurczak, 2020-01-12:
Components of sensor_msgs/PointField are displayed in inconsistent order:

[image]

dirk-thomas, 2020-01-12:
Please provide exact steps how to reproduce the problem.

pauljurczak, 2020-01-12:
The steps are fairly complex, unfortunately. Follow the steps from tutorial at: https://bitbucket.org/osrf/subt/wiki/tutorials/SystemSetupInstall up to the last command in step #5. Replace it with `ign launch -v 4 urban_circuit.ign worldName:=urban_qual robotName1:=X1 robotConfig1:=COSTAR_HUSKY_SENSOR_CONFIG_1`

It may be easier to achieve the same result by using Docker: https://bitbucket.org/osrf/subt/wiki/tutorials/SystemSetupDockerhub.

dirk-thomas, 2020-01-12:
Can you please also show the output your get  from `rostopic echo /X1/points` (in the case where the UI shows inconsistent order).

pauljurczak, 2020-01-13:
`rostopic echo /X1/points/fields` produces correctly ordered listing:
```
- 
  name: "x"
  offset: 0
  datatype: 7
  count: 1
- 
  name: "y"
  offset: 4
  datatype: 7
  count: 1
- 
  name: "z"
  offset: 8
  datatype: 7
  count: 1
- 
  name: "intensity"
  offset: 16
  datatype: 7
  count: 1
- 
  name: "ring"
  offset: 24
  datatype: 4
  count: 1
---
```
BTW, `rqt` completely chokes on `sensor_msgs/PointCloud2` display. I have 6-core modern Intel CPU and the core `rqt` runs on goes to 100% utilization at 4.7GHz. User interface becomes unresponsive and takes about 30s to refresh when `/X1/points` is selected. I'm running Ubuntu 18.04.3 and ROS Melodic.