ros-visualization/rqt #226: [noetic] Cannot launch rqt_reconfigure standalone

doisyg, 2020-05-27:
Loading the plugin from rqt works, however, starting the tool directly with `rosrun rqt_reconfigure rqt_reconfigure ` crashes:

> Traceback (most recent call last):
>   File "/opt/ros/noetic/lib/rqt_reconfigure/rqt_reconfigure", line 5, in <module>
>     from rqt_gui.main import Main
>   File "/opt/ros/noetic/lib/python3/dist-packages/rqt_gui/main.py", line 40, in <module>
>     from rospkg.rospack import RosPack
> ImportError: No module named rospkg.rospack
>

cottsay, 2020-05-27:
You should be able to work around this issue by ensuring that the Python 3 module `rospkg` is installed.

doisyg, 2020-05-28:
Solved it by using python alternative and be sure that `$ python -V` outputs `Python 3.8.2`

PR #227 (2020-06-11): add missing dependencies for rqt_gui: rospkg-modules, python_qt_binding, rospy
Since the `noetic` branch is also the development target branch for `melodic`, we'll need to use a conditional to support both Python versions. Support for conditionals was added in package manifest version 3, so we also need to bump from the version 1 to the version 3 specification.

Here is where rospkg is imported from `rqt_gui`:
https://github.com/ros-visualization/rqt/blob/fd404459cb844f5999c3924f077ff1cb2e5571ef/rqt_gui/src/rqt_gui/main.py#L40

Python 2 rosdep key: [python-rospkg-modules](https://github.com/ro [... cut]
closed 2020-06-11 by dirk-thomas