ros-visualization/rqt #235: Python3 Syntax issue?

rowandempster, 2020-11-19:
When I run `rqt --command-start-plugin rqt_plot.plot.Plot` on ROS Noetic I'm getting
```
Traceback (most recent call last):
  File "/opt/ros/noetic/bin/rqt", line 13, in <module>
    sys.exit(main.main())
  File "/opt/ros/noetic/lib/python3/dist-packages/rqt_gui/main.py", line 61, in main
    return super(
  File "/opt/ros/noetic/lib/python3/dist-packages/qt_gui/main.py", line 385, in main
    (rc, msg) = remote_interface.start_plugin(
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 72, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 141, in __call__
    return self._connection.call_blocking(self._named_service,
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 652, in call_blocking
    reply_message = self.send_message_with_reply_and_block(
dbus.exceptions.DBusException: org.freedesktop.DBus.Python.TypeError: Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/service.py", line 711, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/opt/ros/noetic/lib/python3/dist-packages/qt_gui/plugin_manager_dbus_interface.py", line 59, in start_plugin
    plugin_id = plugins.keys()[0]
TypeError: 'dict_keys' object is not subscriptable
```
Googled and looks like the problem is `dict.keys()` isn't indexable in python3. https://stackoverflow.com/questions/26394748/nltk-python-error-typeerror-dict-keys-object-is-not-subscriptable

rowandempster, 2020-11-19:
@sloretz @dirk-thomas Related to the Noetic release maybe? https://github.com/ros-visualization/rqt/issues/220

dirk-thomas, 2020-11-24:
@rowandempster Neither of the two mentioned people are maintainers of this repository (anymore). @mjeronimo FYI.

mjeronimo, 2020-11-25:
@rowandempster Thanks for reporting. I've reproduced and will submit a fix.

sloretz, 2025-06-02:
ROS 1 is end-of-life (EOL) as May 31st 2025, so I'm closing this.

[If you still rely on ROS 1, read this page to learn about your options.](https://bit.ly/NoeticEOL)