ros-visualization/rqt #238: rqt noetic fails to start in *K*ubuntu 20.04.1

dyle71, 2020-11-26:
Just tried to run rqt, but if fails:

```bash
$ rqt
Traceback (most recent call last):
  File "/opt/ros/noetic/bin/rqt", line 13, in <module>
    sys.exit(main.main())
  File "/opt/ros/noetic/lib/python3/dist-packages/rqt_gui/main.py", line 61, in main
    return super(
  File "/opt/ros/noetic/lib/python3/dist-packages/qt_gui/main.py", line 407, in main
    from python_qt_binding import QT_BINDING
  File "/opt/ros/noetic/lib/python3/dist-packages/python_qt_binding/__init__.py", line 55, in <module>
    from .binding_helper import loadUi, QT_BINDING, QT_BINDING_MODULES, QT_BINDING_VERSION  # @UnusedImport
  File "/opt/ros/noetic/lib/python3/dist-packages/python_qt_binding/binding_helper.py", line 274, in <module>
    _select_qt_binding(
  File "/opt/ros/noetic/lib/python3/dist-packages/python_qt_binding/binding_helper.py", line 108, in _select_qt_binding
    QT_BINDING_VERSION = binding_loader(required_modules, optional_modules)
  File "/opt/ros/noetic/lib/python3/dist-packages/python_qt_binding/binding_helper.py", line 155, in _load_pyqt
    _named_optional_import('PyQt5.%s' % module_name)
  File "/opt/ros/noetic/lib/python3/dist-packages/python_qt_binding/binding_helper.py", line 142, in _named_optional_import
    _named_import(name)
  File "/opt/ros/noetic/lib/python3/dist-packages/python_qt_binding/binding_helper.py", line 133, in _named_import
    module = builtins.__import__(name)
ValueError: PyCapsule_GetPointer called with incorrect name
```

This is a *Kubuntu 20.04.1* (hence the 'K').

```bash
$ lsb_release -a
LSB Version:    core-11.1.0ubuntu2-noarch:printing-11.1.0ubuntu2-noarch:security-11.1.0ubuntu2-noarch
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.1 LTS
Release:        20.04
Codename:       focal
```
Running the very same rqt inside a *Ubuntu 20.04.1* works.

I then omitted QWebKitWidgets, which IIRC has been removed from Qt.

```bash
$ cat -n /opt/ros/noetic/lib/python3/dist-packages/python_qt_binding/binding_helper.py | grep -C 2 ' 95'
    93          'QtWebEngineCore',
    94          'QtWebEngineWidgets',
    95          # 'QtWebKitWidgets',  # Qt 5.0 - 5.5
    96          'QtWebSockets',
    97          'QtX11Extras',
```

This now launches a GUI and I get a dialog, which is empty. But a look at the command line does not make you feel comfy:

```bash
$ rqt
RosPluginProvider.load(qt_gui_cpp/CppPluginProvider) exception raised in __builtin__.__import__(qt_gui_cpp.cpp_plugin_provider, [CppPluginProvider]):
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rqt_gui/ros_plugin_provider.py", line 79, in load
    module = __builtin__.__import__(
  File "/opt/ros/noetic/lib/python3/dist-packages/qt_gui_cpp/cpp_plugin_provider.py", line 33, in <module>
    from .cpp_binding_helper import qt_gui_cpp
  File "/opt/ros/noetic/lib/python3/dist-packages/qt_gui_cpp/cpp_binding_helper.py", line 43, in <module>
    from . import libqt_gui_cpp_sip
ValueError: PyCapsule_GetPointer called with incorrect name

RecursivePluginProvider.discover() loading plugin "qt_gui_cpp/CppPluginProvider" failed:
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/qt_gui/recursive_plugin_provider.py", line 60, in discover
    instance = self._plugin_provider.load(plugin_descriptor.plugin_id(), None)
  File "/opt/ros/noetic/lib/python3/dist-packages/rqt_gui/ros_plugin_provider.py", line 90, in load
    raise e
  File "/opt/ros/noetic/lib/python3/dist-packages/rqt_gui/ros_plugin_provider.py", line 79, in load
    module = __builtin__.__import__(
  File "/opt/ros/noetic/lib/python3/dist-packages/qt_gui_cpp/cpp_plugin_provider.py", line 33, in <module>
    from .cpp_binding_helper import qt_gui_cpp
  File "/opt/ros/noetic/lib/python3/dist-packages/qt_gui_cpp/cpp_binding_helper.py", line 43, in <module>
    from . import libqt_gui_cpp_sip
ValueError: PyCapsule_GetPointer called with incorrect name
```

and any further click on a menu item (e.g. "Help/About") will then bring just more of the same, without any response on th eGUI

```bash
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/qt_gui/about_handler.py", line 62, in show
    from qt_gui_cpp.cpp_binding_helper import qt_gui_cpp
  File "/opt/ros/noetic/lib/python3/dist-packages/qt_gui_cpp/cpp_binding_helper.py", line 43, in <module>
    from . import libqt_gui_cpp_sip
ValueError: PyCapsule_GetPointer called with incorrect name
```

The PyQt sip shipped within ros now links to my Kubuntu installed Qt5 installation:
```bash
$ ldd /opt/ros/noetic/lib/python3/dist-packages/qt_gui_cpp/libqt_gui_cpp_sip.so | grep libQt5Core
        libQt5Core.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Core.so.5 (0x00007fa56d8b1000)
```

Which is Qt 5.12.8
```bash
$ dpkg --list | grep libqt5core
ii  libqt5core5a:amd64                            5.12.8+dfsg-0ubuntu1                        amd64        Qt 5 core module
```

I got to check the one on pure, native Ubuntu, but... if my local Qt Version crashes with the Python SIP you ship, why not have your own libQt5 published within ROS and fiddle LD_PRELOAD path right before running rqt in order to pick that Qt5 lib you compiled against?

... or is there any other remedy for me... >.<

dyle71, 2020-11-26:
Hm, nope. Ubuntu 20.04.1 ships the very same libQt5core version.

```bash
$ dpkg --list | grep libqt5core
ii  libqt5core5a:amd64                          5.12.8+dfsg-0ubuntu1                amd64        Qt 5 core module
```
This is not the cause of the trouble.

Meehhh, ... o.O

julled, 2021-01-23:
Same problem here, running noetic in ubuntu 20.04 dockerized environment.

timonegk, 2021-03-21:
I experienced the same problem in combination with PyKDL which is imported in tf2_geometry_msgs. I reported the bug on the [PyQt mailing list](http://python.6.x6.nabble.com/PyKDL-fails-to-import-after-PyQt5-QtCore-td5288940.html). They pointed out that the problem is that PyKDL is a SIP v4 package and PyQt uses a newer SIP version. The current [master of PyKDL](https://github.com/orocos/orocos_kinematics_dynamics) no longer uses SIP, so for me the problem was solved by uninstalling my system's kdl packages and cloning the kdl repository into my catkin workspace. I don't know if the underlying issue for you is also with PyKDL, but even if it isn't, the problem might be fixed by importing the problematic package before `PyQt5.QtCore` or by compiling a newer version.

rickstaa, 2022-03-16:
For me, this issue was caused by the fact that I manually installed a newer version of QT and pyqt. Removing these installations and re-installing ROS noetic solved the problem.