ros-visualization/rqt #244: ROS2: Publish an array with rqt

giwrgosg, 2021-02-09:
Hi everyone, I have been using rqt to publish messages to test my applications, however I still haven't found a way to publish a message that includes an array.
In my custom message I have a variable size array declared as uint8[]. I notice in rqt that the expected type is "sequence<uint8>".
I have tried different expressions, but couldn't figure out the correct one. Could someone provide an example please?

RyanDelGizzi, 2021-03-13:
Hi @giwrgosg, 

I created the original question on [ROS Answers](https://answers.ros.org/question/370355/publishing-arrays-with-rqt_publisher-in-ros2/) and set out to resolve the issue. Same issue [here](https://github.com/ros-visualization/rqt_publisher/issues/23) for rqt_publisher.

One problem is how [get_slot_types](https://github.com/ros-visualization/rqt/blob/d06bacf1b5d7c356189a5f35e3dc0e20bd30643c/rqt_py_common/src/rqt_py_common/topic_helpers.py#L152) handles array message types.  Since Crystal, the string for array type has changed, using a `<>` syntax instead of a `[]` syntax. And such, the parsing of that info needs to change.

I have a fix for this repo, and also one for [https://github.com/ros-visualization/rqt_publisher/tree/dashing-devel](rqt_publisher) which allows us to publish arrays. However, @mlautman has a (massive) PR #224 that addresses this issue along with a myriad of other crystal->foxy differences. I don't think it makes sense for me to submit a PR on this repo to address a small change that is already addressed in another PR. There are still changes that need to be made to rqt_publisher to support the changes in this repo, but they are rather small.

@mlautman I would love to help move this along as well getting rqt_publisher working as well. If there is a way I can assist, please let me know!

ijnek, 2021-05-15:
Hi @RyanDelGizzi 
:+1:  to your suggested fix, this would be quite desirable. Considering #224 hasn't had any updates in a while, it would be great if you could consider putting a PR up with your changes!
Thanks!