ros-visualization/rqt #249: ROS 2 Rolling/Galactic: Can't see labels/icons on many (but not all) buttons

clalancette, 2021-04-12:
While doing testing for the upcoming Galactic release, I noticed that many of the buttons in RQt on my test platforms were empty.  For instance, running `rqt`, and then adding in the `Plugins->Logging->Bag` plugin shows a group of buttons across the top.  But there are no labels or icons on any of them.  They do all have a tooltip, so it is somewhat possible to figure out what they are for, but something is clearly missing.

@cottsay suggested it may have to do with the vendoring of `tango_icons_vendor`, but I have not yet had time to look into it.  @ahcorde, any thoughts here?

sloretz, 2021-04-14:
Examples of missing icons

https://github.com/ros-visualization/rqt_graph/issues/63
https://github.com/ros-visualization/rqt_service_caller/issues/19

clalancette, 2021-04-30:
Fixed by ros-visualization/qt_gui_core#250

closed 2021-04-30 by clalancette