ros-visualization/rqt #252: rqt: command not found.

mzdz, 2021-06-14:
os version: 
     macOS Big Sur (11.04)

I follow this guide:
    https://docs.ros.org/en/galactic/Installation/macOS-Development-Setup.html

example:
   `ros2 run demo_nodes_cpp talker`
   `ros2 run demo_nodes_py listener`
   `ros2 run turtlesim turtlesim_node`
   `ros2 run turtlesim turtle_teleop_key`
all examples  are work, but run 
  ` rqt`
return:
   `zsh: command not found: rqt`

my .zshrc file:
```
...

source $ZSH/oh-my-zsh.sh
source ~/ros2_galactic/install/setup.sh

export ROS_DOMAIN_ID=100
export OPENSSL_ROOT_DIR=$(brew --prefix openssl)
export PATH="/usr/local/sbin:$PATH"
export CMAKE_PREFIX_PATH=$CMAKE_PREFIX_PATH:/usr/local/opt/qt@5
export PATH=$PATH:/usr/local/opt/qt@5/bin
export ROS_PYTHON_VERSION=3.9

...
```

rqt need to be compiled separately?  or how to solve this question?  thanks!

mzdz, 2021-06-14:
i try compiled rqt separately: 
     follow this[ link](https://docs.ros.org/en/galactic/Guides/RQt-Source-Install-MacOS.html?highlight=rqt) and this [link](https://docs.ros.org/en/galactic/Guides/RQt-Source-Install.html) but i run this command:
     `rosdep install --from-paths src --ignore-src --rosdistro galactic -y --skip-keys "console_bridge fastcdr fastrtps libopensplice67 rti-connext-dds-5.3.1 urdfdom_headers`

it return:

  ```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
osrf_pycommon: No definition of [python3-mock] for OS [osx]
ros2topic: No definition of [python3-numpy] for OS [osx]
examples_rclpy_pointcloud_publisher: No definition of [python3-numpy] for OS [osx]
rosidl_runtime_py: No definition of [python3-numpy] for OS [osx]
rosidl_cli: No definition of [python3-importlib-metadata] for OS [osx]
qt_gui: No definition of [python3-qt5-bindings] for OS [osx]
ros2pkg: No definition of [python3-importlib-resources] for OS [osx]
python_qt_binding: No definition of [python3-qt5-bindings] for OS [osx]
rosidl_parser: No definition of [python3-lark-parser] for OS [osx]
google_benchmark_vendor: No definition of [benchmark] for OS [osx]
...
```

iormaoro, 2021-07-09:
Same problem here.
I can't find a way to install rqt separately.

clalancette, 2021-07-09:
You'll notice that the installation instructions specifically have:
```
colcon build --symlink-install --packages-skip-by-dep python_qt_binding
```

And a note:
```
Note: due to an unresolved issue with SIP, Qt@5, and PyQt5, we need to disable python_qt_binding to have the build succeed. This will be removed when the issue is resolved, see: https://github.com/ros-visualization/python_qt_binding/issues/103
```

The end result of that is that no `rqt` packages are currently compiled on macOS.  Since macOS isn't officially supported in Galactic, the ROS 2 core team won't spend any time on this in the near future.  That said, if you'd like to contribute to https://github.com/ros-visualization/python_qt_binding/issues/103 or the larger tracking issue at https://github.com/ros2/ros2/issues/1148, we are happy to review contributions.

iormaoro, 2021-07-09:
Thanks for that @clalancette . I went and procede with what's suggested in [this comment](https://github.com/ros-visualization/python_qt_binding/issues/103#issuecomment-829247901).
I removed qt_gui_cpp and rqt_gui_cpp in ros-visualization and it went through, I'm now able to play with rqt and procede with the tutorials... it will come back at some point though probably, I will have an eye on the thread. Not able to contribute right now due to lack of knowledge but hopefully in a short future.