ros-visualization/rqt #255: ROS 2: rqt plot not working if any message class can't be loaded

alsora, 2021-07-14:
Hi, I'm running `rqt` with ROS 2 Foxy, installed on Ubuntu 20.04 machine.

If I open rqt-plot, I get the following logs
```
[INFO] [1626281248.362572382] [get_message_class]: Failed to load class: test/msg/TestMsg
TopicCompleter.update_topics(): could not get message class for topic type "test/msg/TestMsg" on topic "/test"
[ERROR] [1626281248.414186684] [get_message_class]: Malformed msg message_type: test/action/DockServo_FeedbackMessage
TopicCompleter.update_topics(): could not get message class for topic type "test/action/DockServo_FeedbackMessage" on topic "/dock/_action/feedback"
[ERROR] [1626281248.429233743] [get_message_class]: Malformed msg message_type: test/action/Undock_FeedbackMessage
TopicCompleter.update_topics(): could not get message class for topic type "test/action/Undock_FeedbackMessage" on topic "/undock/_action/feedback"
[ERROR] [1626281248.430252980] [get_message_class]: Malformed msg message_type: test/action/WallFollow_FeedbackMessage
TopicCompleter.update_topics(): could not get message class for topic type "test/action/WallFollow_FeedbackMessage" on topic "/wall_follow/_action/feedback"
```

I also have many other topics in the system for which there are no errors/warnings.
The presence of the aforementioned logs prevents me from visualizing anything with rqt plot (even the "good topics").
The only way I have for visualizing something is to first stop all the publishers for which a log message is produced.

Why is it required for rpt plot to know all the visible message types in order to work?
This seems counter-intuitive.

I'm purposefully not providing the `test/msg/TestMsg` implementation to the system, so the log is correct about not being able to load it.
On the other hand, I don't understand the error about malformed action messages.
You can find here the 3 actions that are causing it.
If they are really malformed, shouldn't the build step fail?

DockServo.action
```
# Request
---
# Result
bool is_docked
---
# Feedback
bool sees_dock
```

Undock.action
```
# Request
---
# Result
bool is_docked
---
# Feedback
```

WallFollow.action
```
# Request
int8 FOLLOW_RIGHT = -1
int8 FOLLOW_LEFT = 1

int8 follow_side
builtin_interfaces/Duration max_runtime
---
# Result
builtin_interfaces/Duration runtime
---
# Feedback
bool engaged
```

clalancette, 2021-07-14:
> If they are really malformed, shouldn't the build step fail?

I think this is just a red herring.  That message is coming from https://github.com/ros-visualization/rqt/blob/c5f2d8082c3ac2ff14a358c10e386cf2056a38f0/rqt_py_common/src/rqt_py_common/message_helpers.py#L235 , where it is clearly expecting a "message", not an "action".  It's not that the action is malformed, it is that the code is currently expecting that to be a message type, but it is instead an action type.  The real bug may be that it should allow some of the actions through, since actions rely on topics for certain things.  But I'm not certain of that; I'd have to dig deeper to figure out what is really going on there.

alsora, 2021-07-30:
That makes sense.
However, that's only part of the problem.

I have removed all action servers from my system and indeed now I don't see that "malformed message error".
However, I still can't use rqt and my guess is that this is due to the presence of the following log
```
[INFO] [1626281248.362572382] [get_message_class]: Failed to load class: test/msg/TestMsg
TopicCompleter.update_topics(): could not get message class for topic type "test/msg/TestMsg" on topic "/test"
```

The machine that is running rqt does not have access to the IDL definition of TestMsg (this is on purpose), but this prevents me from visualizing any other topic coming from the robot (that on the other hand I have the IDL definition for).
I also noticed that in the rqt plot window the dropdown list of topics to display also contains those ones for which a message is not found. 
However, if I do click on them, rqt crashes with the following error

```
Traceback (most recent call last):
  File "/opt/ros/galactic/lib/python3.8/site-packages/rqt_plot/plot_widget.py", line 278, in on_topic_edit_textChanged
    plottable, message = is_plottable(self._node, topic_name)
  File "/opt/ros/galactic/lib/python3.8/site-packages/rqt_plot/plot_widget.py", line 173, in is_plottable
    fields, message = get_plot_fields(node, topic_name)
  File "/opt/ros/galactic/lib/python3.8/site-packages/rqt_plot/plot_widget.py", line 96, in get_plot_fields
    message_class = get_message(topic_type_str)
  File "/opt/ros/galactic/lib/python3.8/site-packages/rosidl_runtime_py/utilities.py", line 28, in get_message
    interface = import_message_from_namespaced_type(get_message_namespaced_type(identifier))
  File "/opt/ros/galactic/lib/python3.8/site-packages/rosidl_runtime_py/import_message.py", line 30, in import_message_from_namespaced_type
    module = importlib.import_module(
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 961, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 973, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'test'

```

kshfire, 2022-06-03:
I have encountered a similar problem when I executed rqt w/ ros 2 control demo. Thus, the rqt plot function is not working now..

[ERROR]: Malformed msg message_type: control_msgs/action/FollowJointTrajectory_FeedbackMessage
TopicCompleter.update_topics(): could not get message class for topic type "control_msgs/action/FollowJointTrajectory_FeedbackMessage" on topic "/position_trajectory_controller/follow_joint_trajectory/_action/feedback"

Could anyone tell me how to fix this issue ? :)

slyandsmart, 2023-12-13:
Hello, 
i am using ROS2 galactic on a docker container withing ubuntu 20.04 in WSL2. 
I still have the same Error. I can use rqt fine with standard messages but not with custom messages. 

```
[ERROR] [1702488695.788901111] [get_message_class]: Malformed msg message_type: dozer_interfaces/action/BladeAngle_FeedbackMessage
TopicCompleter.update_topics(): could not get message class for topic type "dozer_interfaces/action/BladeAngle_FeedbackMessage" on topic "/blade_pitch_angle/_action/feedback"
TopicCompleter.update_topics(): could not get message class for topic type "dozer_interfaces/msg/MachineInputs" on topic "/drive_clearance/machine_inputs"
```

The problem with the action i can understand. But whats wrong with the dozer_interface/msg/MachineInputs message. It is in the CMakeLists but can not be found by rqt??