ros-visualization/rqt #257: rqt is broken via apt or source

FPSychotic, 2021-12-13:
if I install rqt foxy via apt it is broken.
if I install it via source give a error when execute starting by 
`rqt  line 11, in <module>load_entry_point('rqt-gui==1.1.2`

it was workinf 2 weeks ago with no problem

clalancette, 2021-12-13:
Can you give additional details on the platform that you are on, the version of rqt that you have installed, and the error message that you are getting?  It seems to work OK for me here, with version 1.1.2-1focal.20211112.192306

FPSychotic, 2021-12-13:
> Can you give additional details on the platform that you are on, the version of rqt that you have installed, and the error message that you are getting? It seems to work OK for me here, with version 1.1.2-1focal.20211112.192306

@clalancette 
sorry by the lack of details. it happens with rqt, rqt_graph at least, in noetic and foxy both in 20.04. it was workimg. found a similar error related with qt5 , not sure. 
`mother@imother:~$ rqt
Traceback (most recent call last):
  File "/opt/ros/noetic/bin/rqt", line 13, in <module>
    sys.exit(main.main())
  File "/opt/ros/noetic/lib/python3/dist-packages/rqt_gui/main.py", line 61, in main
    return super(
  File "/opt/ros/noetic/lib/python3/dist-packages/qt_gui/main.py", line 407, in main
    from python_qt_binding import QT_BINDING
  File "/opt/ros/noetic/lib/python3/dist-packages/python_qt_binding/__init__.py", line 55, in <module>
    from .binding_helper import loadUi, QT_BINDING, QT_BINDING_MODULES, QT_BINDING_VERSION  # @UnusedImport
  File "/opt/ros/noetic/lib/python3/dist-packages/python_qt_binding/binding_helper.py", line 274, in <module>
    _select_qt_binding(
  File "/opt/ros/noetic/lib/python3/dist-packages/python_qt_binding/binding_helper.py", line 108, in _select_qt_binding
    QT_BINDING_VERSION = binding_loader(required_modules, optional_modules)
  File "/opt/ros/noetic/lib/python3/dist-packages/python_qt_binding/binding_helper.py", line 155, in _load_pyqt
    _named_optional_import('PyQt5.%s' % module_name)
  File "/opt/ros/noetic/lib/python3/dist-packages/python_qt_binding/binding_helper.py", line 142, in _named_optional_import
    _named_import(name)
  File "/opt/ros/noetic/lib/python3/dist-packages/python_qt_binding/binding_helper.py", line 133, in _named_import
    module = builtins.__import__(name)
ValueError: PyCapsule_GetPointer called with incorrect name
imother@imother:~$ rqt_graph 
Traceback (most recent call last):
  File "/opt/ros/noetic/bin/rqt_graph", line 8, in <module>
    sys.exit(main.main(sys.argv, standalone='rqt_graph.ros_graph.RosGraph'))
  File "/opt/ros/noetic/lib/python3/dist-packages/rqt_gui/main.py", line 61, in main
    return super(
  File "/opt/ros/noetic/lib/python3/dist-packages/qt_gui/main.py", line 407, in main
    from python_qt_binding import QT_BINDING
  File "/opt/ros/noetic/lib/python3/dist-packages/python_qt_binding/__init__.py", line 55, in <module>
    from .binding_helper import loadUi, QT_BINDING, QT_BINDING_MODULES, QT_BINDING_VERSION  # @UnusedImport
  File "/opt/ros/noetic/lib/python3/dist-packages/python_qt_binding/binding_helper.py", line 274, in <module>
    _select_qt_binding(
  File "/opt/ros/noetic/lib/python3/dist-packages/python_qt_binding/binding_helper.py", line 108, in _select_qt_binding
    QT_BINDING_VERSION = binding_loader(required_modules, optional_modules)
  File "/opt/ros/noetic/lib/python3/dist-packages/python_qt_binding/binding_helper.py", line 155, in _load_pyqt
    _named_optional_import('PyQt5.%s' % module_name)
  File "/opt/ros/noetic/lib/python3/dist-packages/python_qt_binding/binding_helper.py", line 142, in _named_optional_import
    _named_import(name)
  File "/opt/ros/noetic/lib/python3/dist-packages/python_qt_binding/binding_helper.py", line 133, in _named_import
    module = builtins.__import__(name)
ValueError: PyCapsule_GetPointer called with incorrect name
imother@imother:~$ 
`

clalancette, 2021-12-13:
So is the problem you are having with Foxy or Noetic?  The original comment said Foxy, but the above shows a backtrace from Noetic.

FPSychotic, 2021-12-13:
I have installed  both, and happens in both.In this case I had sourced
noetic, but if I source ros2 foxy same issue, first noted it in Foxy, were
it was working.

clalancette, 2021-12-13:
I'm not seeing it with Foxy.  Are you happening to source *both* Foxy and Noetic at the same time?  That could potentially cause issues.

FPSychotic, 2021-12-14:
> I'm not seeing it with Foxy. Are you happening to source _both_ Foxy and Noetic at the same time? That could potentially cause issues.

Yes, I had both sourced, but later O sourced them individually, and same error. Now I have uninstalled foxy, and the error persist

FPSychotic, 2021-12-14:
solved with this
pip3 uninstall PyQt5-sip 
pip3 install PyQt5
https://answers.ros.org/question/371691/valueerror-pycapsule_getpointer-called-with-incorrect-name/

thank you so much by be so supportive and fast. thanks by share and maintain this code, people like me still would be in the stone era without people like you, sharing knowledge and work.

closed 2021-12-14 by FPSychotic

LoyVanBeek, 2021-12-30:
For me, it worked to both
```bash
pip3 uninstall PyQt5-sip
pip3 uninstall PyQt5
```
:man_shrugging: