ros-visualization/rqt #266: Qt6

ijnek, 2022-03-30:
Is anyone working on updating Qt5 to Qt6 on rolling?

As briefly mentioned on [Qt for Python 6.2 announcement](https://www.qt.io/blog/qt-for-python-release-6.2) under the heading *Support for Python 3.10*,  there were some changes required to get Qt for Python working with Python3.10 and these changes are only in pyside6.

There's an issue that arrises with using Python3.10 with PyQt5 (ror py (and Qt5 in general), where implicit conversion from float to int  doesn't happen. See for more: https://bugs.python.org/issue36048

The best solution seems to upgrade to Qt6 for this exact problem.

clalancette, 2022-03-30:
As far as I know, nobody is working on the upgrade right now.  If you'd like to contribute, we'd be happy to review, though note that we still have to maintain PyQt5 compatibility to support Ubuntu 20.04 for now.

It would also be worthwhile to test out RQt as it stands right now and see if this problem is happening.  If so, a detailed report of the problem would be worthwhile so we can see if there are other mitigations we can apply.

ijnek, 2022-03-31:
@clalancette I see the problem happing, and I've added a [report of the problem to rqt_bag](https://github.com/ros-visualization/rqt_bag/issues/108), where the error actually happens.

ijnek, 2022-04-25:
It seems that corresponding repositories are adapting to PyQt5 and mitigating these issues.
Qt6 - I see this happening further down the road and I don't think tracking progress in this issue is appropriate, so I'm going to close this issue.

closed 2022-04-25 by ijnek