ros-visualization/rqt #270: `ros2 run rqt rqt` No Executable Found

sloretz, 2022-05-13:
```
C:\Users\IEUser>ros2 run rqt rqt
No executable found
```

`rqt` works fine, but `ros2 run rqt rqt` does not. For that to work, the executable needs to be installed twice like `rviz` does

https://github.com/ros2/rviz/blob/7b4ba855b5803ef8b77b7732889c8b1b6e43532b/rviz2/CMakeLists.txt#L82-L83

I don't know how to do that with a `setup.py`. Most ROS 2 packages use the `setup.cfg` to change the install location, but it's not obvious to me how to duplicate that.

https://github.com/ros-visualization/rqt/blob/302a2ea7f9a2f85c4e2d11d793419d576d47fa15/rqt_gui/setup.py#L41

Voldivh, 2023-05-11:
After taking a closer look, I just noticed that the correct command we have to use is `ros2 run rqt_gui rqt_gui` and we get the exact same behavior as when we only run the entry point `rqt`. I'm not sure if this is the expected behavior, I think it does because on the package `rqt` we don't have any executable or UI or anything, it is basically description for the package.

Voldivh, 2023-05-11:
Let me know what you think @clalancette

clalancette, 2023-05-15:
> After taking a closer look, I just noticed that the correct command we have to use is `ros2 run rqt_gui rqt_gui` and we get the exact same behavior as when we only run the entry point `rqt`. I'm not [...]

If I take a look at https://github.com/ros-visualization/rqt/blob/95d3d6e4335ad93244fb4063ce2fa2fe9e80e532/rqt_gui/setup.py, then I see two things:

```
    data_files=[
        ('share/' + package_name + '/resource', [
            'resource/rqt.graffle',
            'resource/rqt.png',
            'resource/rqt.svg',
        ]),
        ('share/ament_index/resource_index/packages',
            ['resource/' + package_name]),
        ('share/' + package_name, ['package.xml']),
        ('lib/' + package_name, ['bin/rqt_gui'])
    ],
```

and

```
    entry_points={
        'console_scripts': [
            'rqt = rqt_gui.main:main',
        ],
    },
```

What I think is happening here is that the `rqt` binary that we can run is getting installed by the `console_scripts`, and the `ros2 run rqt_gui rqt_gui` that we can run is getting installed as one of the `data_files`.

What I think we should do here is to fix it so that we can run either `rqt` or `ros2 run rqt_gui rqt`.  To do that, I suggest the following:
1.  Add in a setup.cfg file to this package that is exactly the same as the one e.g. in https://github.com/ros2/geometry2/blob/rolling/tf2_tools/setup.cfg .  That means that when we go to install the `rqt` entry point, we'll install it to `lib` (so we'll find that entry point when doing `ros2 run rqt_gui rqt`).
2. Edit the `data_files` installation so that we install `bin/rqt` into `bin/rqt`.  That means that we'll still be able to run just a plain `rqt`.
3. Update https://github.com/audrow/yatm/blob/a4164cd6dfcf5c85ad2a258e9d304671f2a7404d/requirements/features-executable.yaml#L513 to say `ros2 run rqt_gui rqt`.#291 

@Voldivh Does that make sense?  Can you work on that update?

Voldivh, 2023-05-15:
> @Voldivh Does that make sense? Can you work on that update?

Yes, I think it all makes sense. I'll start working on it.