ros-visualization/rqt #314: Qt6 Migration

oursland, 2024-09-18:
Qt5 and PySide2 are no longer being supported on some platforms.  Nor are these libraries compatible with Python 3.12.  There should be a plan for migrating to Qt6 and PySide6 to ensure continued functionality of `rqt`.

clalancette, 2024-09-21:
> Nor are these libraries compatible with Python 3.12.

That doesn't seem to be the case; we are successfully using Qt5 and PySide2 with Python 3.12 on Ubuntu 24.04.

> There should be a plan for migrating to Qt6 and PySide6 to ensure continued functionality of `rqt`.

But agreed.  The last time I looked into this, it wasn't actually possible to do on Ubuntu 24.04 because it didn't have the newer version of PySide packaged.

oursland, 2024-09-21:
Indeed, there does exist ports to Python 3.12.  My information came from the conda-forge porting effort to Python 3.12, which was abandoned when RedHat abandoned their efforts.  Since then I've been informed that Debian and other distributions have completed the port, while others have dropped Qt5 from their repositories altogether.

The official support for Qt5 ended on May 26, 2023, the subscribers-only extended support for Qt 5.15 will continue until May 26, 2025.