ros-visualization/rqt #317: [Notice] Nav2 Migrated to TwistStamped to replace Twist for cmd_vel Topics

SteveMacenski, 2024-12-03:
Hi,

As a key community project using Nav2, I wanted to drop you a note that I am in the process of changing the default cmd_vel topic type from Twist to TwistStamped in efforts of enabling a broader set of applications and safety constraints. This has been an on-going discussion over the last 18 months and now has the technology stack in GZ, ROS Control, and Nav2 to enable it.

To migrate, your robot and simulation should soon accept a TwistStamped message rather than a Twist message. We have a couple of provided utilities that could make this easier for you, by switching to use a Twist Publisher and Twist Subscriber [1-2]. This is what is now used in all Nav2 nodes to either publish or subscribe to a Twist or TwistStamped message.

If you would like to continue to use a Twist, you can set enable_stamped_cmd_vel to false in all nodes in your configuration that use Twist Publisher/Subscriber objects, but this will be shortly switched to being default true.

[1] https://github.com/ros-navigation/navigation2/blob/main/nav2_util/include/nav2_util/twist_publisher.hpp
[2] https://github.com/ros-navigation/navigation2/blob/main/nav2_util/include/nav2_util/twist_subscriber.hpp

Nav2 migration:
- https://github.com/ros-navigation/navigation2/pull/4779
- https://github.com/ros-navigation/docs.nav2.org/pull/614
- https://github.com/ros-navigation/nav2_minimal_turtlebot_simulation/pull/16

SteveMacenski, 2024-12-03:
I'm not 100% sure if any updates to RQT are needed. Can a maintainer on this metaproject give me some guidance? 

I only see [rqt_robot_steering](https://github.com/ros-visualization/rqt_robot_steering), but only in ROS 1 (but there is a branch for stamped https://github.com/ros-visualization/rqt_robot_steering/commit/ad65f939abd75404dc55d92ae26c8cc516d9f05a)

clalancette, 2024-12-03:
As far as I can tell, there are no users of Twist in rqt other than rviz (which is the ROS 1 version), and rqt_robot_steering.  So I'm going to close this one out, and suggest that you open an individual issue on rqt_robot_steering, which does indeed have a ROS 2 port.

closed 2024-12-03 by clalancette